f4exb/sdrangel #1959: Duplicates of "Description" appearing in Configuration Presets each time the app is launched

Axpelle, 2024-01-24:
I have noticed that each new description added by me in Configuration Presets is duplicated over and over each time I launch SDRA. I'm using the latest version on Sonoma 14.2.1 

Screenshot is attached.

Any suggestion as to why this happens?

[Duplicates in Presets.pdf](https://github.com/f4exb/sdrangel/files/14036760/Duplicates.in.Presets.pdf)

srcejon, 2024-01-24:
Press the update button 
[image]
 instead of add?

Axpelle, 2024-01-24:
> From "srcejon" ***@***.***>
> To "f4exb/sdrangel" ***@***.***>
> Cc "Axpelle" ***@***.***>; "Author"
> Date 24/01/2024, 14:22:57
> Subject Re: [f4exb/sdrangel] Duplicates of "Description" appearing in
> Configuration Presets each time the app is launched (Issue #1959)

I have meticulously deleted all the presets and started with a clean 
sheet using a new preset. Yet on relaunching multiple copies of 
previously deleted presets reappear all by themselves. This issue seems 
to apply only to user-defined presets.

Very baffling.

Axpelle, 2024-01-24:
Deleted the app entirely and did a fresh installation. Seems to have solved the issue.

closed 2024-01-27 by f4exb