f4exb/sdrangel #1962: Is there support for SDRAngel <-> SDRangel via TCP?

lattice0, 2024-01-26:
The reverse API supports some stuff like providing an API for others to control SDRangel.

Is there a way to access my computer's SDRAngel through my tablet SDRAngel?

srcejon, 2024-01-26:
Can SDRangel control all the features of another remote SDRangel - no.

But if you have an SDR attached on your computer that is running SDRangel, that you want to stream IQ data to process on SDRangel running on your tablet, then you can use: 

   Remote TCP Sink on PC: https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/remotetcpsink/readme.md
   Remote TCP Input on Tablet: https://github.com/f4exb/sdrangel/blob/master/plugins/samplesource/remotetcpinput/readme.md

lattice0, 2024-01-26:
Thanks it worked!

closed 2024-01-27 by f4exb