f4exb/sdrangel #1969: Crashing undoes all of the settings

LingKuja, 2024-01-30:
There's still an issue with the current version of the program, where if by any chance it crashes, it will revert all of the config to the last time it was closed safely, this means that all of the presets and markers will be cancelled from the program's memory.
A fix I would suggest is that the program would periodically save everything, a bit like how SDR++ and SDR# do as well. Or otherwise a way of saving the workspace and all markers, presets and stuff manually via button or hotkey like CTRL+S.

srcejon, 2024-01-30:
You can save all of the settings via the configuration dialog (the star button on the toolbar).

Autosaving defaults should be a straightforward enough addition though, I guess.

thomasgi1, 2024-02-11:
> You can save all of the settings via the configuration dialog (the star button on the toolbar).

No, that star button does not save all the settings on disk. When clicking on the '+' button in the 'Configurations' dialog, it creates the new configuration only in memory despite the tooltip, which says: 'Save current workspaces configuration as new configuration'. The new configurations are only saved to disk once the program is successfully closed. This is a very annoying behaviour. Today I lost 10 new configurations in this way due to a crash.

LingKuja, 2024-02-11:
@thomasgi1 I think @srcejon meant the star button on the device window, I was puzzled too at first.
Honestly I hope more SDR softwares start having more demodulators and stuff built in instead of relying on third party plugins or external software, i would love for a progam with sdr console's ui and sdrangel's usefulness.
[image]

thomasgi1, 2024-02-11:
> @thomasgi1 I think @srcejon meant the star button on the device window, I was puzzled too at first.
The star button on the device window has the same odd behaviour as the start button on the workspace. It does not save to disk until the application is closed. So, we have two save buttons, which do not, what their tooltip suggests. If one wants to save settings to disk, he must close the application.

closed 2024-03-29 by github-actions[bot]