f4exb/sdrangel #1970: Problem with Sdrangel Spectrum

Valstee, 2024-02-02:
Hi,
My project is to try to make satellite receptions in S bands. 
I would like to put in the head of the antenna a Raspberry, with a LNA and a SDR. 
The RPi is powered by PoE, so I just have a network cable that comes out of the installation. 
But I'd like to get some spectrum feedback for testing purposes....

My problem is the following, I used to use Sdrangel spectrum but for some time now I can't get it to work..., the build goes fine (ie. no errors) but it doesn't work, and I don't see any errors, neither on sdrangelspectrum, nor in the logs of sdrangel.... 
Can you reproduce the problem? Does it work for you?

Thanks, 
F4JIZ

f4exb, 2024-03-04:
Do not forget to start the spectrum server with a POST to the endpoint /sdrangel/deviceset/{deviceSetIndex}/spectrum/server

Valstee, 2024-03-07:
I've just solved my problem. I had a problem launching the server. I had identified this problem: 

`(I) WSSpectrum::openSocket: cannot start spectrum server at 127.0.0.1 on port 8887`

Which was actually linked to an ip address problem between the local address, the network address and the container address...