f4exb/sdrangel #1972: F-Droid Support

aadityabhatia, 2024-02-03:
Please consider publishing this app to F-Droid, one of the most popular app stores for FOSS apps built from source. It will increase the reach and popularity of the app by making it available to users / devices not on Google ecosystem. See [F-Droid Quick Start Guide](https://f-droid.org/docs/Submitting_to_F-Droid_Quick_Start_Guide/) for details.

Davem734, 2024-02-14:
I second this request. I have a de-googled tablet running LineageOS and would prefer to install SDRAngel from F-Droid to have notifications and easy access to updates. 

And thank you all for the hard work that has gone into this amazing project...

aadityabhatia, 2024-03-26:
What inactivity?

Davem734, 2024-04-16:
A bump to keep the dream alive (for the bot's sake)

closed 2024-06-02 by github-actions[bot]