f4exb/sdrangel #1974: Cannot list USRP device

keukenrol, 2024-02-05:
Hello

I have followed the guide describing how to get samples from an USRP device.
uhd_find_devices lists the following (serial not shown):
addr: 192.168.10.2
fpga: HG
name:
product: X310
type: x300
Under device user arguments I then set the device: 
[image]
However, once restarted the device is not listed in the sampling devices (from add RX device).
Any idea?

I am able to connect using GNU Radio.

Thanks!

srcejon, 2024-02-05:
Possibly no one has tried an X300 before. Have a look in the log file and see if it says anything relevant.

keukenrol, 2024-02-06:
(D) DeviceUSRP::enumOriginDevices: Could not find any USRP device

srcejon, 2024-02-06:
Can you post anything it says before that as well please?

keukenrol, 2024-02-06:
[sdrangel.log](https://github.com/f4exb/sdrangel/files/14177219/sdrangel.log)

srcejon, 2024-02-06:
Try ip_addr instead of addr=

keukenrol, 2024-02-06:
No difference, both with and without brackets.

keukenrol, 2024-02-06:
Edit: the log file told me that the FPGA firmware is incompatible.
2024-02-06 15:42:10.737 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.testsource
2024-02-06 15:42:10.737 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.usrp
2024-02-06 15:42:12.958 (D) DeviceUSRP::enumOriginDevices: found USRP device "[0:$1] CENSORED"
2024-02-06 15:42:12.958 (D) DeviceUSRPParams::open: type=x300,addr=192.168.10.2,fpga=HG,name=,serial=CENSORED,product=X310
2024-02-06 15:42:14.069 (D) DeviceUSRPParams::open: exception: RuntimeError: Expected FPGA compatibility number 38, but got 39:

The FPGA image on your device is not compatible with this host code build.
Download the appropriate FPGA images for this version of UHD.
As an Administrator, please run:

"C:/Program Files\SDRangel\uhd\utils\uhd_images_downloader.py"

Thanks!

srcejon, 2024-02-23:
UHD now being used in 7.18.1

closed 2024-02-23 by srcejon

Ali6114, 2024-02-27:
Currently, I have set the below settings. Fortunately, SDRangel doesn't crash but the spectrum & Waterfall updating/processing becomes slow.
[image]