f4exb/sdrangel #1975: Link between AIS demodulator and raw signal ?

theoallouche, 2024-02-06:
Hi,

I have an .sdrqiq file and I'm using the AIS demodulator on it.

I'd like to be able to link the detected messages to the associated IQs, or at least have somehow a timestamp that could link the messages to the raw signal.

How could I do that?
Thanks

srcejon, 2024-02-06:
At the moment, the AIS timestamp is only the computers time, so you'd have to adjust the timestamps by hand from when you press start. You could possibly do this by importing the csv file in to Excel.

(The satellite tracker can get it's time from the the IQ file, but I think that's the only plugin that does currently).

srcejon, 2024-03-03:
Have just added a button that allows the choice between having packet date & times come from system clock or File Input device.

closed 2024-03-03 by srcejon