f4exb/sdrangel #1978: Audio glitch in DAB Demod due to Integer overflow

FLDev-net, 2024-02-09:
With loud DAB audio programs, i have some glitches in the decoded audio. Seems to be an integer overflow, when samples are fullscale. 
[image]

srcejon, 2024-02-09:
What's the volume setting?

Screenshot would be useful

srcejon, 2024-02-09:
One thing to check is if there is any resampling going on. What is the DAB stream sample rate? And are those waveforms from the demod analyzer?

Do you mean to have a frequency delta of 108kHz? What do you get if you set that to 0Hz?

FLDev-net, 2024-02-09:
Yes, waveforms are from Demod Analyser, making unpleasant cracks in audio.  
With 0Hz offset, I have the same peaks. 

These peaks are visible only with stations with fullscale audio. 
Sample rate is 48k 

For example, Skyrock in France always have a loud sound processing. 
[image]

[image]
[image]

Have a nice day.

srcejon, 2024-02-09:
Do you see the problem with other DAB decoders?

Can you capture and attach an IQ file that has the problem?

Thanks

FLDev-net, 2024-02-09:
Hi, 

File is too big to upload here. 
Here is the URL to download it : 

https://www.swisstransfer.com/d/7fb87351-cea3-4d1d-a729-ba7221a0ee60

It's a 800MB sdriq file recorded from the frequency 181,936Mhz (DAB 6A), and packed in a zip file. 
The station with the most loud audio is SKYROCK (Id 61981).

Radio station in France usually broadcast at full scale. No headroom for audio. So, some samples reach the limit of Int32 when they are decoded. 

Normally, we expect this could lead to a slight saturation, but not an integer overflow effect as I have seen. 

You asked if on other software I have the same effect, and the answer is no. 

I wish you a good day.

closed 2024-03-27 by github-actions[bot]

jfdelnero, 2024-08-01:
Hello,

i have the same issue, in France / Paris too, 

https://hxc2001.com/vrac/DAB_Paris_6C_Nostalgie_1_Aug_2024_wav.7z

[image]

i will try to dump the audio buffer right after the AAC decoder.

jfdelnero, 2024-08-01:
For sure the issue is related to the audio interpolation / decimation.
i managed to "fix" the issue by bypassing the resampling code :

```
diff --git a/plugins/channelrx/demoddab/dabdemodsink.cpp b/plugins/channelrx/demoddab/dabdemodsink.cpp
index 86d9e0d8a..30ae8a8b5 100644
--- a/plugins/channelrx/demoddab/dabdemodsink.cpp
+++ b/plugins/channelrx/demoddab/dabdemodsink.cpp
@@ -403,6 +403,9 @@ void DABDemodSink::audio(int16_t *buffer, int size, int samplerate, bool stereo)
             ci.real(0.0f);
             ci.imag(0.0f);
         }
+
+        processOneAudioSample(ci);
+#if 0
         if (m_audioInterpolatorDistance < 1.0f) // interpolate
         {
             while (!m_audioInterpolator.interpolate(&m_audioInterpolatorDistanceRemain, ci, &ca))
@@ -419,6 +422,7 @@ void DABDemodSink::audio(int16_t *buffer, int size, int samplerate, bool stereo)
                 m_audioInterpolatorDistanceRemain += m_audioInterpolatorDistance;
             }
         }
+#endif
     }
 }
```

Indeed, don't forget to set the audio output sample rate to 48000 Hz.

@FLDev-net  @srcejon

srcejon, 2024-08-01:
Thanks. What were the two sample rates?

jfdelnero, 2024-08-02:
> Thanks. What were the two sample rates?

The issue appeared with up-sampling (48 kHz -> 96 kHz) and down-sampling (48kHz -> 24 kHz), and even without resampling (48 kHz -> 48 kHz), but this not a surprise since interpolation and decimation use the same polyphase filter function. 

Anyway i found the root cause of the issue and made a fix and a pull request.

(see 
https://github.com/jfdelnero/sdrangel/commit/2fddaff6d24d64c9afe88cf97b7ea6848d674c83
)