f4exb/sdrangel #1979: MacOS: GUI blocked after accessing the Audio Device Selection dialog

thomasgi1, 2024-02-11:
Once the Audio Device Selection dialog (right click on 'Mute/unmute audio' button) was opened and then closed the GUI is no longer usable.  Clicking on either the 'Mute/unmute audio', the 'Start al devices in workspace' or 'Configuration presets' button has no effect. The only way to recover from that situation is to restart the application.

UncleBuck62, 2024-02-16:
This also occurs if you right click the calibration button in the spectrum display.

Also, if you right click on the Websocket Spectrum settings, it opens two copies of the same dialog and the same lockup issues happen as described above.

srcejon, 2024-02-28:
Seems to be specific to Mac. It's a problem on both Qt 5 and 6.

srcejon, 2024-03-11:
Have noticed on Linux dialogs can sometimes appear twice. Possibly related to gestures, as I see:

    QGestureManager::deliverEvent: could not find the target for gesture

On Ubuntu 22.04, when the dialog doesn't open twice. On 23.10, we don't get this message, but and the dialog opens twice.

srcejon, 2024-03-12:
Looks like there's two separate issues:

- Dialogs opening twice. This should be fixed by the above patch.
- Open Config button being non-responsive after a dialog has been opened. Not sure what the problem is here.

srcejon, 2024-03-13:
Reopened as only partially fixed so far.

mattsoftware, 2025-04-28:
Just a note to say I think I just hit this too, trying to edit audio, right click to select which audio device to use for a specific demod, and its behaving like there is a dialog somewhere but nothing is visible on screen.

PR #2017 (2024-03-12): 2 Bug Fixes
Prevent dialogs from being opened twice on Linux and Mac when right clicking. Partially fixes #1979

Hide ADS-B maps that aren't available. For #2010