f4exb/sdrangel #1982: Auto DC Culling dip in frequency

LingKuja, 2024-02-12:
[image]
In short, my issue is that for some reason if I enable the Auto DC culling there's a dip around the center, where the antenna should be located. Any fix? This didn't happened with other programs like SDRConsole (but it doesn't have all the demodulators like SDRAngel lol)

srcejon, 2024-02-13:
> my issue is that for some reason if I enable the Auto DC culling there's a dip around the center,

That's exactly what that option is supposed to do.

> Any fix? 

You don't set the device centre frequency to the frequency you are trying to receive, you set it lower / higher, then use the delta f option in the SSB settings (that is hidden in your picture) to tune to the frequency you are interested in.

> This didn't happened with other programs like SDRConsole 

Haven't used it for a while, but any DC removal feature is going to do the same. Perhaps SDRconsole is automatically applying a frequency offset or something.

LingKuja, 2024-02-13:
> You don't set the device centre frequency to the frequency you are trying to receive, you set it lower / higher, then use the delta f option in the SSB settings (that is hidden in your picture) to tune to the frequency you are interested in.

That is what I've been doing, I've just noticed this dip, and used the SSB demodulator to highlight it. Good to know that it's the intended behavior.

f4exb, 2024-02-13:
Welcome to SDR Zero IF technology...

closed 2024-03-31 by github-actions[bot]