f4exb/sdrangel #1991: Both Mac versions have no SDRplay plugin loaded

loewal, 2024-02-20:
As the title says: v.7.18.0 has no supprt for SDRplay.
The latest that worked was v7.17.3

Axpelle, 2024-02-20:
It's working fine with my RSP1A on Mac mini running Sonoma with M2 chip and SDRPlay 3.14 installed.

srcejon, 2024-02-20:
For 7.18, you need to make sure you have SDRplay API v3.14 installed.

loewal, 2024-02-20:
Thanks Jon, that worked!
But...2 sideeffects:
1) ADSB: map is blank
2) SDR++ has no SDRplay source anymore; sorry for this, not your concern I realize...

srcejon, 2024-02-20:
> 1. ADSB: map is blank

This is expected with the arm64 version for now, as it doesn't work with QT6. It should work on x64 version, as that is still built using QT5.

> SDR++ has no SDRplay source anymore; sorry for this, not your concern I realize...

That will need to be updated to use the latest SDRplay API as well. We need to use the latest in order to support the new RSP1B.

closed 2024-02-20 by srcejon

loewal, 2024-02-23:
Why can I add the Map feature, which works well and not the map in ADSB?
[image]

srcejon, 2024-02-23:
Because I rewrote the Map to support Qt6, but haven't had time to do ADS-B yet.