f4exb/sdrangel #1992: Hardware Selftest RTL-SDR V4 ? | ADS-B did not receive data

ds2k5, 2024-02-21:
Hi,
got a RTL-SDR V4

```
$ lsusb
Bus 002 Device 008: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

```

Using Version: 7.17.3 of SDRAngel

[output.log](https://github.com/f4exb/sdrangel/files/14356304/output.log)

Using channel:   ADS-B Demoulator

But did not get Data

Is there a way to check that the RTL-SDR is working or is defective ?

Did build a antenna:  8,5 Thigh Dipole

The Cable looks ok if I check with a multimeter, there is no short circuit.

[image]

srcejon, 2024-02-21:
Make the spectrum window bigger and zoom in on the signal a bit. (Start by pressing the A button, to auto-scale)

Try receiving in FM band.

Are you near many aircraft? Get the antenna outside.

ds2k5, 2024-02-21:
@srcejon 
thanks for you help

>Try receiving in FM band.

how to enable audio to head if the radio station is recived fine

[image]

ds2k5, 2024-02-21:
[image]

Is there a comand line tool or something else I can quick test if
the radio station a 92.2 MHz is recived...

my I have a issue with the sdlangel    it is self compiled because did not find a package

srcejon, 2024-02-21:
Turn the gain up.

I'd say it's unlikely to be a SDRangel compilation problem, given what the screenshots show.

ds2k5, 2024-02-21:
>Turn the gain up.

AGC enabled is not good ?

srcejon, 2024-02-21:
> AGC enabled is not good ?

It only enables AGC in one stage of the RTL SDR, not all.

> got only noise but no radio signal

Try another antenna.

ds2k5, 2024-02-21:
>Try another antenna.

no change

could this be a library issue ?

ds2k5, 2024-02-21:
/usr/local/bin$ ./rtl_test 
Found 1 device(s):
  0:  RTLSDRBlog, Blog V4, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...

ds2k5, 2024-02-21:
tryed another SDR Software (SDR++) 
can listen to Radio Station, so I belive the Hardware is OK
so I need to find out why the sdrangel is not working...

try to build new with the libsdr from RTL-SDR Blog

srcejon, 2024-02-21:
Could be your driver doesn't support V4.

In the log you sent, I see: 

[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!
[R82XX] PLL not locked!

srcejon, 2024-02-21:
> Do i need to include: https://github.com/rtlsdrblog/rtl-sdr-blog insted of the RTL-SDR from the build-manual ? https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

In the build instructions, we have:

    git clone https://github.com/osmocom/rtl-sdr.git librtlsdr
    cd librtlsdr
    git reset --hard be1d1206bfb6e6c41f7d91b20b77e20f929fa6a7

And it seems that tag is from 2019, so will not support V4.

I would suggest rebuilding that library with the latest version.

srcejon, 2024-02-21:
@f4exb - Shall we bump that to 420086af84d7eaaf98ff948cd11fea2cae71734a - which is release 2.0.1 and should include V4 support?

ds2k5, 2024-02-21:
thanks to @srcejon @f4exb 

did a new build with the  libsdr from RTL-SDR Blog

now it works!

closed 2024-02-21 by ds2k5

f4exb, 2024-02-21:
> @f4exb - Shall we bump that to 420086af84d7eaaf98ff948cd11fea2cae71734a - which is release 2.0.1 and should include V4 support?

Yes it seems OK