f4exb/sdrangel #1993: High CPU usage in sdrangel (over 100%) - ADS-B - how to fix ?

ds2k5, 2024-02-21:
Hi,
have Debian 12 x86_64

Compiled the sdrangel
when running inactive the CPU Load is around 

But when running ADS-B
CPU Load is over 100% by one CPU

[image]

[image]

[image]

Any hint for Compiling options that not have so high CPU load?

When using  Boradcast FM Demodulator CPU load is around  42 %  for sdrangel process

srcejon, 2024-02-21:
ADS-B demod requires significantly more CPU than FM. ADS-B demod runs at 4MSa/s, whereas FM is running at 48kHz. It's like comparing broadband to dial-up.

You can possibly reducing the load by adjusting sample rates and threshold, but this will affect packet error rate.

ds2k5, 2024-02-21:
@srcejon 

>You can possibly reducing the load by adjusting sample rates and threshold, but this will affect packet error rate.

thank you

closed 2024-04-08 by github-actions[bot]