f4exb/sdrangel #1997: Compilation error

CocolinoFan, 2024-02-25:
**Problem**
When attempting to install the SDRAngel software package on Gentoo Linux using the emerge command, the installation process fails during the compilation phase of the dev-libs/cm256cc package. The failure is due to compilation errors related to missing type definitions (uint16_t, uint32_t, uint8_t) in the data.h header file.

**Expected Behavior**
The dev-libs/cm256cc package should compile successfully without errors, enabling the installation of the SDRAngel software.

**Steps to Reproduce**
1. Attempt to install SDRAngel software on Gentoo Linux using the emerge command ([available through an overlay](https://gpo.zugaina.org/net-wireless/sdrangel))
2. Observe the compilation failure of the dev-libs/cm256cc package during the build process.

**Additional Information**
SDRAngel version: 7.18.1
Gentoo Linux version: Kernel: 6.7.4-gentoo-dist
Relevant logs and configurations: [build log](https://pastebin.com/KxvEYbs0) | [emerge --info =dev-libs/cm256cc-1.1.0-r1::hamari](https://pastebin.com/ExgHu3Zv)

**Proposed Solution**
The compilation errors in the data.h header file of the dev-libs/cm256cc package seem to stem from missing type definitions. Check if `#include <cstdint>` is included in [example0.cpp](https://github.com/f4exb/cm256cc/blob/master/unit_test/example0.cpp)

srcejon, 2024-02-28:
data.h includes stdint.h, which should define these types - and I can compile the code successfully on gentoo 6.6.* from by cloning the repo.

The question is, why isn't stdint.h defining those types for you?

What's in your /usr/include/stdint.h and /usr/include/bits/stdint-uint.h

CocolinoFan, 2024-02-28:
Ok, I am embarrassed to admit but I don't know much programming. I am more then willing to help how ever I can, but I have no idea what is going on in such a big program like SDRAngel.
Here are my `stdint.h` and `stdint-uint.h` (`stdint-uintn.h`):
[files.zip](https://github.com/f4exb/sdrangel/files/14438662/files.zip)

P.S. I think my file is called stdint-uint**n**.h for whatever reason.

srcejon, 2024-02-28:
Sorry, stdint-uintn.h is the correct name.

I can see in there the definition:

   typedef __uint16_t uint16_t;

That your build.log says is missing.

Perhaps the compiler picks up the file from somewhere else. Can you copy and paste the contents of CMakeFiles/cm256_tx.dir/unit_test/example0.cpp.o.d from under the build directory, please?

CocolinoFan, 2024-02-28:
I can not find that file >...<
Here is my whole build directory 
[buildDir.zip](https://github.com/f4exb/sdrangel/files/14439054/buildDir.zip)

srcejon, 2024-02-29:
> I can not find that file 

You are building using ninja rather than make. It appears ninja deletes the .d files generated by gcc, and puts them in .ninja_deps. The dependencies can then be viewed with `ninja -t deps`

However, the dependencies are missing in this file for the file that is failing compilation (example0.cpp), so we can't see what it's reading using that method.

It would be useful if you could add two additional flags to the compiler, -H and -save-temps. I'm not sure how you do that with emerge, but running cmake from the command line, you'd do:

    cmake -E env CXXFLAGS="-H -save-temps" cmake -G Ninja ..
    ninja

If you could then please send the console output and build directory, we can see what it's doing.

Alternatively, you might be able to set emerge to use make rather than ninja, to see if the .d file is created.

Unfortunately, I I've tried running Gentoo in a VM to reproduce, bit it keeps crashing.

CocolinoFan, 2024-02-29:
That's odd, the Gentoo Gods strike me as the kind of people that would use make, even if ninja is faster. I wander if it's the [overlay](https://gpo.zugaina.org/net-wireless/sdrangel) that is using ninja for some reason 🤔.

In any case, I added `CXXFLAGS="-H -save-temps"` in my `/etc/portage/make.conf` and I think that did the trick. Now, searching for `find . -name "*example0*"` gave me some results:
`./cm256cc-1.1.0-r1/work/cm256cc-1.1.0_build/CMakeFiles/cm256_tx.dir/unit_test/example0.cpp.ii`
`./cm256cc-1.1.0-r1/work/cm256cc-1.1.0_build/CMakeFiles/cm256_tx.dir/unit_test/example0.cpp.s`
`./cm256cc-1.1.0-r1/work/cm256cc-1.1.0/unit_test/example0.cpp`
`./cm256cc-1.1.0-r1/work/cm256cc-1.1.0/unit_test/example0.h`

Here is the new build directory: [buildDir.zip](https://github.com/f4exb/sdrangel/files/14446710/buildDir.zip)
And here is the console output: [output.txt](https://github.com/f4exb/sdrangel/files/14446818/output.txt)

srcejon, 2024-02-29:
Ok, it seems the SDRangel compilation instructions haven't been updated to use the latest version of cm256cc that fixes the problem when compiling with gcc 13. I've now updated them.

For the Gentoo build, it seems the package management files pick up the latest cm256cc release tarball via this Manifest file: https://data.gpo.zugaina.org/hamari/dev-libs/cm256cc/Manifest

However, the latest cm256cc (release)[https://github.com/f4exb/cm256cc/releases] is a few years old and so doesn't include this fix. So I guess it would be useful if @f4exb could create a new release of cm256cc using the latest source, then you'll have to get in touch with whoever is maintaining the Gentoo stuff to update it and hopefully it should compile.

f4exb, 2024-02-29:
> So I guess it would be useful if @f4exb could create a new release of cm256cc using the latest source

Done: https://github.com/f4exb/cm256cc/releases/tag/v1.1.1

CocolinoFan, 2024-02-29:
Thank you sir. I will get into contact with the owner of the overlay and get SDRAngel working again on Gentoo.

CocolinoFan, 2024-02-29:
Works like a charm. Thank you so much for your work 🙏 
Works beautifully on Gentoo now. Time to surf the spectrum 
[image]

closed 2024-02-29 by CocolinoFan