f4exb/sdrangel #2002: Update HackRF Compilation Instructions

CRD716, 2024-02-29:
Currently the wiki says to use v2022.09.01, but this version is two years out of date. The most recent version is [v2024.02.1](https://github.com/greatscottgadgets/hackrf/releases/tag/v2024.02.1). It may also be worth checking through the new documentation to look for possible improvements/features.

CRD716, 2024-02-29:
Another note: the old v2022 version used means that the last two hardware revisions (r9 and r10) are incompatible.

CRD716, 2024-02-29:
I remember attempting on the previous release and it seemed fine, will check with the new one when I have the chance.

CRD716, 2024-03-29:
Some notes:

- https://github.com/f4exb/libsigmf is 5 commits behind, is there any reason why we shouldn't try to be compatible with upstream?
- https://github.com/drowe67/codec2-dev, why are we using such an old version in the instructions? It seems to work fine at the latest.

Most importantly, the latest hackrf version works good, works with new firmware too (which means the new revisions should work too), still think a look over for new features/improvements may be warranted.

srcejon, 2024-03-29:
Both of the upstream versions of those projects don't build on Windows.

CRD716, 2024-04-15:
dsdcc has also been updated, works fine.

CRD716, 2024-05-26:
@f4exb could at least the hackrf version be updated?

CRD716, 2024-07-06:
Is there something I'm missing? This should be a one line change, right?

closed 2025-06-23 by srcejon