f4exb/sdrangel #2010: Map feature ADSB

loewal, 2024-03-09:
Hi Jon,
Since there is support for Qt6, does that mean that SDRangel Arm version supports the map with ADSB?

Regards

loewal, 2024-03-09:
> Hi Jon, Since there is support for Qt6, does that mean that SDRangel Arm version supports the map with ADSB?
> 
> Regards

Sorry...add: I tried it, but still white surface.

srcejon, 2024-03-09:
It should have fixed it, but I see an error:

    Qt5Compat.GraphicalEffects is not installed.

So I'll have to see why this isn't installed on the mac.

srcejon, 2024-03-09:
New build uploaded

loewal, 2024-03-09:
Yes! That did it Jon.
Thanks very much for the superquick solution.
Regards

loewal, 2024-03-09:
Sorry Jon, I was to quick:
- x64 version: selec mapboxgl --> crash;  osm-map okay with centering on "My position"
- Arm version: select mapboxgl --> white surface; nocrash; no centering screen on "My Position" using osm-map, but visible map

Regards

srcejon, 2024-03-11:
Qt 6 on Mac doesn't support mapboxgl, so I'd expect that.

srcejon, 2024-03-12:
I'll update the list, so only available plugins are listed.

loewal, 2024-04-16:
Hi Jon,
I installed version 7.20.0 Arm (no Intel yet?)
Using ADSB, the map is centered near London and not my position.
As I mentioned before, using the Intel version, the map is okay.
Could you please look into this?
Tanks!

closed 2024-04-16 by f4exb