f4exb/sdrangel #2011: (Info) Compilation on Pi5 won’t boot on Pi4

alphafox02, 2024-03-09:
I’m not exactly sure why, but compiling SDRangel on a Pi5 runs, but same sd card then inserted in a Pi4 leads to an abort when trying to start SDRangel. 

Doing everything the other way around, compiling on a Pi4 first, SDRAngel then will start in both versions of the Pi. This is the only application I’ve seen so far exhibit that behavior. 

Maybe some kind of additional cpu flag set on a Pi5?

srcejon, 2024-03-09:
Most likely. ARM v8.2 vs ARM v8.

alphafox02, 2024-03-09:
I’ll close this out. That’s probably it. Just wasn’t sure what was unique in the case of SDRAngel compiling that was picking up the difference. I’ll go back and look but there was a few applications, like SDR++ and Satdump that I compiled this way.

I think it’s probably best I compile things on the Pi4 if I then intend to use the sd card on the Pi5 since it seems doing it that way keeps compatibility.

closed 2024-03-09 by alphafox02

alphafox02, 2025-09-24:
This actually is still a problem today, but it’s the only application I’ve seen this happen with. I’m just curious, what part of SDRAngel is so specific to that small change? 

I suppose nothing can be done about it or is there something that can be passed to the build process on a pi5 that will allow it to boot on a 4? It just takes so long to build it on a 4. If there’s a way to cross compile it on a desktop in order to use it on arm v8, that would be even better.