f4exb/sdrangel #2018: Code Rate settings for chirpchat demodulator

tomtommyshh, 2024-03-12:
I’m currently working with a LoRa device and using AT commands to configure it for peer-to-peer (P2P) communication. I’ve set the frequency, spreading factor, bandwidth, preamble length, and payload length using the at+set_config=lorap2p command. However, I’m unsure about the code rate setting in the chirpchat demodulator settings.

Here’s a snippet of my code for reference:

`sendCommand("at+set_config=lorap2p:433000000:7:0:1:8:20\r\n");`

Could anyone guide me on how to set the code rate for the chirpchat demodulator settings in SDRangel? Any help would be greatly appreciated!

Thank you in advance

f4exb, 2024-03-13:
You have first to check with your application what do these parameters mean in terms of code rate. Then you have to adjust the bandwidth (https://github.com/f4exb/sdrangel/blob/master/plugins/channelrx/demodchirpchat/readme.md#3-bandwidth) to match the code rate.

f4exb, 2024-03-22:
Be careful that usually with a spread factor (SF) of 11 or 12 comes a "distance enhancement" (DE) factor of 2

closed 2024-05-07 by github-actions[bot]