f4exb/sdrangel #2022: MacOS: no spectrum

Czukuru, 2024-03-16:
MacOs Sonoma. The is no spectrum, i got some kind of screen mirroring. If i move windows in program, in spectrum they will  move too. I tried different releases and same issue.
[image]
[image]

srcejon, 2024-03-16:
Look in log file (preferences > logging menu) to see if that tells you anything.

Czukuru, 2024-03-16:
> Look in log file (preferences > logging menu) to see if that tells you anything.

Czukuru, 2024-03-16:
> Look in log file (preferences > logging menu) to see if that tells you anything.

[log — копия.txt](https://github.com/f4exb/sdrangel/files/14622756/log.txt)

Czukuru, 2024-03-16:
> > Look in log file (preferences > logging menu) to see if that tells you anything.
> 
> [log — копия.txt](https://github.com/f4exb/sdrangel/files/14622756/log.txt)
> 
> is it right file?

[image]

Czukuru, 2024-03-16:
if i move cursor around this window, i can change frequency by dragging something that should be in spectrum. so computer thinks that he is spectrum there but wrong visual output

srcejon, 2024-03-16:
It's an opengl problem of some sort.

Try running from a terminal to see if anything is output there.

Czukuru, 2024-03-16:
> It's an opengl problem of some sort.
> 
> Try running from a terminal to see if anything is output there.

can you explain a little please. im not really sure what to do. i run this app from terminal, but no difference

srcejon, 2024-03-16:
I was just hoping some error message might be output - but apparently not.

Czukuru, 2024-03-16:
So the is no possible chance to fix this it in my case?

Czukuru, 2024-03-16:
> Я просто надеялся, что появится какое-нибудь сообщение об ошибке, но, видимо, нет.

I saw some people who got the same problem, but no solution. Can i do something to fix this?

Czukuru, 2024-03-16:
Installing from macports or from dmg, or different versions won't help

Czukuru, 2024-03-16:
[image]

srcejon, 2024-03-17:
If you're able to compile the source yourself, you could try changing sdrgui/gui/glspectrumview.cpp. On line 900, change:

    // Enable OpenGL debugging
    // Disable for release, as some OpenGL drivers are quite verbose and output
    // info on every frame
    if (false)

To 

    // Enable OpenGL debugging
    // Disable for release, as some OpenGL drivers are quite verbose and output
    // info on every frame
    if (true)

And it might give us some information about the problem.

srcejon, 2024-03-17:
> I saw some people who got the same problem, but no solution. Can i do something to fix this?

Might be useful if you can link to those some how, so we can see what you have in common. E.g. h/w, s/w etc.

closed 2024-05-02 by github-actions[bot]