f4exb/sdrangel #2023: Need Help with SDRangel Server Setup and interfacing with ADALM-Pluto

imca-potato, 2024-03-17:
Greetings!

I would like to ask how exactly is SDRangel Server setup on the device? Do we need to install something else? I don't quite understand the instructions in the wiki (particularly about the Docker stuff) since I am quite new to Linux systems. Your guidance will be highly appreciated. :>

As for interfacing with the ADALM-Pluto, I manage to make this work on my laptop (Windows 10 OS), but I intend to use the software with a Raspberry Pi 4. When I try to use SDRangel with ADALM-Pluto on the RPi, the Pluto disconnects and I am unable to decode any signals. What may be the problem with this? As additional information, my RPi has an Ubuntu Mate 22.04 OS installed.

Thank you in advance for your response!

srcejon, 2024-03-17:
What do you want it to be setup to do?

imca-potato, 2024-03-17:
I'd want it to demodulate LoRa signals.

imca-potato, 2024-03-18:
I'm expecting something similar to the GUI-based SDRangel, where the data is sent to a local host and is retrieved from there.

I forgot to add that as much as possible, I'd like for the entire application to be run on the command line instead of through GUI, hence why I want to use the SDRangel Server.

imca-potato, 2024-03-21:
Greetings!

I would like to follow this up. :>

imca-potato, 2024-04-18:
Good day!

I would like to ask how to setup the SDRangel such that I can control it via command line. Specifically, I would like to clarify if I first have to install and run the Docker images before I am able to control it via the REST API, which I assume is the command line version of controlling SDR?

srcejon, 2024-04-18:
No, you don't need docker to use the rest API.

I suggest you do Help'About in the GUI then click on the link that takes you to the interactive swagger dcs, which will allow you to try REST commands from your browser.

imca-potato, 2024-04-18:
I see. Thank you for the response! I will check this out.

imca-potato, 2024-04-18:
Greetings! I would like to confirm if the server implementation currently does not support LoRa demodulation?

srcejon, 2024-04-18:
> I seem to be having trouble with setting the appropriate SDR device. I want to be using my ADALM-Pluto which is connected directly to my computer. However, I cannot figure out how to do so. Please help!

You need to give more information about what you are trying to do and what is going wrong. Include text and pictures.

imca-potato, 2024-04-18:
Greetings.

I managed to figure things out. I was using the POST /sdrangel/deviceset command to create an instance of a receiver, but I did not quite figure out how to change the receiver device (it is set to file input by default) at first.

[image]

With a little exploring, I found the command to set the appropriate device.
[image]

Now my only issue is, I am not quite sure how this translates to text that I can execute via command line. How must I go about this?

For instance, if I want to use the above PUT command, how must I input it into my terminal in order for it to execute? For reference I am using Ubuntu OS.

srcejon, 2024-04-18:
If you expand those, it should give you a curl command you can send from a terminal.

imca-potato, 2024-04-18:
I see, do the curl commands need certain packages to be installed?

closed 2024-06-03 by github-actions[bot]