f4exb/sdrangel #2025: v17.9.1 ARM crashes immediately: libsdrgui.dylib missing

loewal, 2024-03-18:
Contents when starting from within SDRangel.app/Contents/MacOS:

What should I do?

dyld[65474]: Library not loaded: @rpath/libsdrgui.dylib
  Referenced from: <27DC3F6F-1540-3875-89C1-6271A9A77ACC> /Applications/Korte_Golf/SDRangel.app/Contents/MacOS/SDRangel
  Reason: tried: '/opt/build/sdrangel/build/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/lib/libsdrgui.dylib' (no such file), '/Users/jon/Qt/6.6.2/macOS/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/jon/Qt/6.6.2/macOS/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/libusb/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/libusb/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/hidapi/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/hidapi/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/opus/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/opus/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/zlib/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/zlib/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/faad/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/faad/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/_deps/boost-build/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/_deps/boost-build/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/x264/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/x264/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/x265/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/x265/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/fdk-aac/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/fdk-aac/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/mpg123/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/mpg123/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/lame/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/lame/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/ffmpeg/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/ffmpeg/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/opencv/src/opencv-build/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/opencv/src/opencv-build/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/codec2/src/codec2-build/src/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/codec2/src/codec2-build/src/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/cm256cc/src/cm256cc-build/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/cm256cc/src/cm256cc-build/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/mbelib/src/mbelib-build/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/mbelib/src/mbelib-build/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/serialdv/src/serialdv-build/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/serialdv/src/serialdv-build/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/dsdcc/src/dsdcc-build/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/dsdcc/src/dsdcc-build/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/apt/src/apt-build/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/apt/src/apt-build/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/sgp4/src/sgp4-build/libsgp4/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/sgp4/src/sgp4-build/libsgp4/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/dab/src/dab-build/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/dab/src/dab-build/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/rtlsdr/src/rtlsdr-build/src/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/rtlsdr/src/rtlsdr-build/src/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/limesuite/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/limesuite/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/soapysdr/src/soapysdr-build/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/soapysdr/src/soapysdr-build/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/airspy/src/airspy-build/src/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/airspy/src/airspy-build/src/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/airspyhf/src/airspyhf-build/libairspyhf/src/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/airspyhf/src/airspyhf-build/libairspyhf/src/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/hackrf/src/hackrf-build/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/hackrf/src/hackrf-build/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/libiio/src/libiio-build/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/libiio/src/libiio-build/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/bladerf/src/bladerf-build/output/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/bladerf/src/bladerf-build/output/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/uhd/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/uhd/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/lib/libsdrgui.dylib' (no such file), '/Users/jon/Qt/6.6.2/macOS/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/Users/jon/Qt/6.6.2/macOS/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/libusb/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/libusb/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/hidapi/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/hidapi/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/opus/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/opus/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/zlib/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/zlib/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/faad/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/faad/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/_deps/boost-build/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/_deps/boost-build/lib/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/bu [... cut ...] ibiio-build/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/bladerf/src/bladerf-build/output/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/bladerf/src/bladerf-build/output/libsdrgui.dylib' (no such file), '/opt/build/sdrangel/build/external/uhd/lib/libsdrgui.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/uhd/lib/libsdrgui.dylib' (no such file)
zsh: abort      ./SDRangel

srcejon, 2024-03-18:
Looks like a load of files are missing from the .dmg - strange, as nothing should have changed - I'll have to rebuild it.

loewal, 2024-03-18:
Curious…..

srcejon, 2024-03-19:
I updated Qt from 6.6.0 to 6.6.2, and it seems that doesn't include the frameworks in the build. Have rolled back to 6.6.0 for now.

Axpelle, 2024-03-19:
I got this on opening:  Cannot mix incompatible Qt library (6.6.0) with this library (6.6.2)
zsh: abort

loewal, 2024-03-19:
This is the crashreport now:
2024-03-19 10:25:44.894 (F) Cannot mix incompatible Qt library (6.6.0) with this library (6.6.2)

loewal, 2024-03-19:
> I got this on opening: Cannot mix incompatible Qt library (6.6.0) with this library (6.6.2) zsh: abort

Sorry for double...

Axpelle, 2024-03-19:
Latest version fixed the Qt incompatibility. That was very quick. Bravo

loewal, 2024-03-19:
Great Jon!
Thanks very much.

closed 2024-03-20 by srcejon