f4exb/sdrangel #2028: APRS igate tries to UTF-8 re-encode APRS packets

hessu, 2024-03-18:
Hi,

I'm Hessu, OH7LZB, I maintain aprs.fi and the aprsc server software which the APRS igate implementation connects to. I briefly reviewed the APRS igate in sdrangel and noticed the following:

APRSWorker::handleMessage tries to re-encode complete APRS packets to UTF-8. While some fields of APRS packets are UTF-8 encoded these days (text message contents, status text, position packet comment text), the complete APRS packets seen in the wild often contain byte sequences which are not valid UTF-8, and the decoding and re-encoding of the complete packet will fail or produce a modified duplicate packet which differs from the original received packet. This is undesirable - the packets need to be handled in binary-clean buffers by igates.

Here's the encoding attempt: https://github.com/f4exb/sdrangel/blob/cb89392c86e4a218c53cbec6b1afeff93cd75f23/plugins/feature/aprs/aprsworker.cpp#L118

Here's the relevant part of my "hints for igate developers" doc:
https://github.com/hessu/aprsc/blob/main/doc/IGATE-HINTS.md#packets-getting-modified-due-to-character-encoding-issues

Thanks!

srcejon, 2024-03-20:
Thanks for the review. Yes, using binary buffer makes sense, so I'll make the change. This will also enable UTF-8 support in the PacketDemod and APRS Feature.

I don't think there's a problem with APRSWorker::send though, as it has a length parameter.

hessu, 2024-03-29:
Thanks for the fixes.

I added a couple of comments in the commit, most importantly it looks for a CR+LF sequence, instead of looking for either a CR or a LF. To get proper protection against command injection the code needs to look for either a CR or an LF, truncate at the first occurrence of either, and always append a CR + LF.

hessu, 2024-03-29:
UTF-8 decoding done still done around the code will probably find on the packets which are not valid UTF-8, but I'm not familiar with QT so I'm not sure what the results will be. There are quite a lot of packets out there with unprintable byte sequences, so debug dumps of packets are best served by a logging method which shows non-ascii bytes as their hexadecimal representations, like "<0x1F><0x1E>".

UTF-8 decoding should generally be done for string fields _after_ APRS decoding, for the text message & comment fields for example. Some valid packets might fail UTF-8 decoding because there's a weird Mic-E position packet byte sequence in there, but then have a valid UTF-8 encoded comment text in the end.

srcejon, 2024-04-14:
Should be fixed in 7.20

closed 2024-04-14 by srcejon