f4exb/sdrangel #2030: sdrangel quit unexpectedly on MacOS M1 , 14.4 (23E214)

Andrej64, 2024-03-20:
This happens when i see "loading plugins". 
Maybe this happens only when there is no SDR device connected( I am considering buying one).

srcejon, 2024-03-20:
You don't need an SDR connected.

Try running from a terminal to see if that gives any more info.

srcejon, 2024-03-20:
Also note, there was a problem with the first mac build of v7.19.1, so make sure you redownload the latest.

Andrej64, 2024-03-20:
I used the latest (yesterday). Terminal:

024-03-20 14:11:04.698 (D) DeviceMetisScan::scan: discovery writeDatagram sent to 192.168.1.255
2024-03-20 14:11:04.698 (D) DeviceMetisScan::scan: discovery writeDatagram sent to 127.255.255.255
2024-03-20 14:11:04.698 (D) DeviceMetisScan::scan: start 0.5 second timeout loop
2024-03-20 14:11:05.175 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplemimo.plutosdrmimo
2024-03-20 14:11:05.175 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplemimo.testmi
2024-03-20 14:11:05.175 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplemimo.testmosync
2024-03-20 14:11:05.175 (D) DeviceEnumerator::addNonDiscoverableDevices: start
2024-03-20 14:11:05.184 (D) MainWindow::MainWindow: load current configuration...
2024-03-20 14:11:05.184 (D) MainWindow::loadConfiguration: configuration [default | no name] 2 workspace(s) - 1 device set(s) - 0 feature(s)
zsh: segmentation fault

2024-03-20 14:11:04.544 (D) PluginManager::loadPluginsDir: fileName: libinputsdrplayv3.dylib
2024-03-20 14:11:04.545 (W) PluginManager::loadPluginsDir: Cannot load library /Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib: (dlopen(/Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib, 0x0085): Library not loaded: /usr/local/lib/libsdrplay_api.so.3.14
  Referenced from: <641EE61B-EA63-3EE0-9B6A-3FD7010A775D> /Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib
  Reason: tried: '/usr/local/lib/libsdrplay_api.so.3.14' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/libsdrplay_api.so.3.14' (no such file), '/usr/local/lib/libsdrplay_api.so.3.14' (no such file))
2024-03-20 14:11:04.545 (I) PluginManager::loadPluginsDir: Soapy SDR disabled skipping libinputsoapysdr.dylib
2024-03-20 14:11:04.545 (D) PluginManager::loadPluginsDir: fileName: libinputtcpremote.dylib
2024-03-20 14:11:04.547 (I) PluginManager::loadPluginsDir: loaded plugin libinputtcpremote.dylib
2024-03-20 14:11:04.547 (D) PluginManager::loadPluginsDir: fileName: libinputtestsource.dylib
2024-03-20 14:11:04.549 (I) PluginManager::loadPluginsDir: loaded plugin libinputtestsource.dylib
2024-03-20 14:11:04.549 (D) PluginManager::loadPluginsDir: fileName: libinputusrp.dylib
2024-03-20 14:11:04.559 (W) PluginManager::loadPluginsDir: Cannot load library /Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputusrp.dylib: (dlopen(/Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputusrp.dylib, 0x0085): Library not loaded: @rpath/Python3.framework/Versions/3.9/Python3
  Referenced from: <4F27982E-0B6E-346C-9A91-6516F2B0840C> /Applications/SDRangel.app/Contents/Frameworks/libuhd.4.6.0.dylib
  Reason: tried: '/opt/build/sdrangel/build/external/uhd/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/uhd/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/libusb/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/libusb/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/hidapi/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/hidapi/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/opus/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/opus/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/zlib/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/zlib/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/faad/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/faad/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/_deps/boost-build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/_deps/boost-build/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/x264/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/x264/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/x265/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/x265/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/fdk-aac/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/fdk-aac/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/mpg123/lib/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/buil [... cut ...] /libsgp4/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/sgp4/src/sgp4-build/libsgp4/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/build/external/dab/src/dab-build/Python3.framework/Versions/3.9/Python3' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/build/sdrangel/build/external/dab/src/dab-build/Python3.framework/Versions/3.9/Python3' (no such file), '/opt/build/sdrangel/bu

Andrej64, 2024-03-20:
still segfault:

3:15.186 (D) PluginManager::loadPluginsDir: fileName: libdemodnavtex.dylib
2024-03-20 14:33:15.188 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnavtex.dylib
2024-03-20 14:33:15.188 (D) PluginManager::loadPluginsDir: fileName: libdemodnfm.dylib
2024-03-20 14:33:15.189 (I) PluginManager::loadPluginsDir: loaded plugin libdemodnfm.dylib
2024-03-20 14:33:15.189 (D) PluginManager::loadPluginsDir: fileName: libdemodpacket.dylib
2024-03-20 14:33:15.191 (I) PluginManager::loadPluginsDir: loaded plugin libdemodpacket.dylib
2024-03-20 14:33:15.191 (D) PluginManager::loadPluginsDir: fileName: libdemodpager.dylib
2024-03-20 14:33:15.192 (I) PluginManager::loadPluginsDir: loaded plugin libdemodpager.dylib
2024-03-20 14:33:15.192 (D) PluginManager::loadPluginsDir: fileName: libdemodradiosonde.dylib
2024-03-20 14:33:15.193 (I) PluginManager::loadPluginsDir: loaded plugin libdemodradiosonde.dylib
2024-03-20 14:33:15.193 (D) PluginManager::loadPluginsDir: fileName: libdemodrtty.dylib
2024-03-20 14:33:15.195 (I) PluginManager::loadPluginsDir: loaded plugin libdemodrtty.dylib
2024-03-20 14:33:15.195 (D) PluginManager::loadPluginsDir: fileName: libdemodssb.dylib
2024-03-20 14:33:15.196 (I) PluginManager::loadPluginsDir: loaded plugin libdemodssb.dylib
2024-03-20 14:33:15.196 (D) PluginManager::loadPluginsDir: fileName: libdemodvor.dylib
2024-03-20 14:33:15.198 (I) PluginManager::loadPluginsDir: loaded plugin libdemodvor.dylib
2024-03-20 14:33:15.198 (D) PluginManager::loadPluginsDir: fileName: libdemodwfm.dylib
2024-03-20 14:33:15.199 (I) PluginManager::loadPluginsDir: loaded plugin libdemodwfm.dylib
2024-03-20 14:33:15.199 (D) PluginManager::loadPluginsDir: fileName: libdoa2.dylib
2024-03-20 14:33:15.200 (I) PluginManager::loadPluginsDir: loaded plugin libdoa2.dylib
2024-03-20 14:33:15.200 (D) PluginManager::loadPluginsDir: fileName: libfeatureafc.dylib
2024-03-20 14:33:15.202 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureafc.dylib
2024-03-20 14:33:15.202 (D) PluginManager::loadPluginsDir: fileName: libfeatureais.dylib
2024-03-20 14:33:15.203 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureais.dylib
2024-03-20 14:33:15.203 (D) PluginManager::loadPluginsDir: fileName: libfeatureambe.dylib
2024-03-20 14:33:15.205 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureambe.dylib
2024-03-20 14:33:15.205 (D) PluginManager::loadPluginsDir: fileName: libfeaturedemodanalyzer.dylib
2024-03-20 14:33:15.206 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturedemodanalyzer.dylib
2024-03-20 14:33:15.206 (D) PluginManager::loadPluginsDir: fileName: libfeaturegs232controller.dylib
2024-03-20 14:33:15.209 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturegs232controller.dylib
2024-03-20 14:33:15.209 (D) PluginManager::loadPluginsDir: fileName: libfeaturelimerfe.dylib
2024-03-20 14:33:15.213 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturelimerfe.dylib
2024-03-20 14:33:15.213 (D) PluginManager::loadPluginsDir: fileName: libfeaturepertester.dylib
2024-03-20 14:33:15.215 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturepertester.dylib
2024-03-20 14:33:15.215 (D) PluginManager::loadPluginsDir: fileName: libfeatureradiosonde.dylib
2024-03-20 14:33:15.216 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureradiosonde.dylib
2024-03-20 14:33:15.216 (D) PluginManager::loadPluginsDir: fileName: libfeatureremotecontrol.dylib
2024-03-20 14:33:15.217 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureremotecontrol.dylib
2024-03-20 14:33:15.217 (D) PluginManager::loadPluginsDir: fileName: libfeaturerigctlserver.dylib
2024-03-20 14:33:15.219 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturerigctlserver.dylib
2024-03-20 14:33:15.219 (D) PluginManager::loadPluginsDir: fileName: libfeaturesatellitetracker.dylib
2024-03-20 14:33:15.221 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturesatellitetracker.dylib
2024-03-20 14:33:15.221 (D) PluginManager::loadPluginsDir: fileName: libfeaturesimpleptt.dylib
2024-03-20 14:33:15.223 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturesimpleptt.dylib
2024-03-20 14:33:15.223 (D) PluginManager::loadPluginsDir: fileName: libfeatureskymap.dylib
2024-03-20 14:33:15.270 (I) PluginManager::loadPluginsDir: loaded plugin libfeatureskymap.dylib
2024-03-20 14:33:15.270 (D) PluginManager::loadPluginsDir: fileName: libfeaturestartracker.dylib
2024-03-20 14:33:15.273 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturestartracker.dylib
2024-03-20 14:33:15.273 (D) PluginManager::loadPluginsDir: fileName: libfilesink.dylib
2024-03-20 14:33:15.275 (I) PluginManager::loadPluginsDir: loaded plugin libfilesink.dylib
2024-03-20 14:33:15.275 (D) PluginManager::loadPluginsDir: fileName: libfilesource.dylib
2024-03-20 14:33:15.276 (I) PluginManager::loadPluginsDir: loaded plugin libfilesource.dylib
2024-03-20 14:33:15.276 (D) PluginManager::loadPluginsDir: fileName: libfreqscanner.dylib
2024-03-20 14:33:15.279 (I) PluginManager::loadPluginsDir: loaded plugin libfreqscanner.dylib
2024-03-20 14:33:15.279 (D) PluginManager::loadPluginsDir: fileName: libfreqtracker.dylib
2024-03-20 14:33:15.280 (I) PluginManager::loadPluginsDir: loaded plugin libfreqtracker.dylib
2024-03-20 14:33:15.280 (D) PluginManager::loadPluginsDir: fileName: libheatmap.dylib
2024-03-20 14:33:15.282 (I) PluginManager::loadPluginsDir: loaded plugin libheatmap.dylib
[... cut ...]
2024-03-20 14:33:16.003 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplemimo.testmi
2024-03-20 14:33:16.003 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplemimo.testmosync
2024-03-20 14:33:16.003 (D) DeviceEnumerator::addNonDiscoverableDevices: start
2024-03-20 14:33:16.011 (D) MainWindow::MainWindow: load current configuration...
2024-03-20 14:33:16.011 (D) MainWindow::loadConfiguration: configuration [default | no name] 2 workspace(s) - 1 device set(s) - 0 feature(s)
zsh: segmentation fault

srcejon, 2024-03-20:
It looks like you've been able to run it successfully before. What SDR were you using?

Try running with --scratch to start with an empty workspace:

    /Applications/SDRangel.app/Contents/MacOS/SDRangel --scratch

And you could also delete:

    /Applications/SDRangel.app/Contents/Resources/lib/plugins/liboutputusrp.dylib

(But try one at a time, so we know if either fixes it).

Andrej64, 2024-03-20:
I tried this: /Applications/SDRangel.app/Contents/MacOS/SDRangel --scratch
and it runs. 
After that i tried: /Applications/SDRangel.app/Contents/MacOS/SDRangel
and it runs with a toolbar at the top of the window. No other windows were showing. Don't know if this is normal. 
BTW, i never had any SDR. 

thanks.

srcejon, 2024-03-20:
If no windows visible, select Workspace > New from the menu.

Andrej64, 2024-03-20:
still no windows.
maybe i should restore 
/Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputsdrplayv3.dylib
/Applications/SDRangel.app/Contents/Resources/lib/plugins/libinputusrp.dylib

srcejon, 2024-03-20:
You shouldn't need either of those files.

Can you post a screenshot please?

Quick start guide is here: https://github.com/f4exb/sdrangel/wiki/Quick-start

closed 2024-05-05 by github-actions[bot]