f4exb/sdrangel #2036: Compile error

EMEJl9l, 2024-03-29:
When attempting to compile sdrangel, the following error occurs:

/opt/build/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp: In constructor ‘DABDemodSink::DABDemodSink(DABDemod*)’:
/opt/build/sdrangel/plugins/channelrx/demoddab/dabdemodsink.cpp:523:30: error: invalid conversion from ‘void (*)(int, void*)’ to ‘tii_data_t’ {aka ‘void (*)(int)’} [-fpermissive]
523 | m_api.tii_data_Handler = tiiDataHandler;
| ^~~~~~~~~~~~~~
| |
| void (*)(int, void*)
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
make[2]: *** [pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/build.make:132: pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/dabdemodsink.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:12964: pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

[image]

srcejon, 2024-03-29:
Possibly due to the version of the DAB library you have. Which version do you have checked out?

EMEJl9l, 2024-03-29:
I followed all the instructions:

cd /opt/build
git clone https://github.com/srcejon/dab-cmdline
cd dab-cmdline/library
git checkout msvc
mkdir build; cd build
cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/opt/install/libdab ..
make -j $(nproc) install

srcejon, 2024-03-29:
What version of SDRangel source do you have?

This looks like: https://github.com/f4exb/sdrangel/issues/1631 or https://github.com/f4exb/sdrangel/issues/1623

EMEJl9l, 2024-03-29:
Apologies for the inconvenience. I recompiled the DAB library, and everything is working now!

closed 2024-03-29 by EMEJl9l