f4exb/sdrangel #2037: ax.25 decoder ignores the H bit in digipeater path

hessu, 2024-03-29:
https://github.com/f4exb/sdrangel/blob/8f33bad363bed4579e073d2554b0f5884f81f25e/sdrbase/util/ax25.cpp#L74

The "via" path decoder ignores the MSB in the last octet, and does not append a '*' in the end of the callsign. As a result, the information whether the packet has traversed the indicated digipeaters, or not, is lost. This breaks "who heard who" type of analysis.

https://www.ax25.net/AX25.2.2-Jul%2098-2.pdf **3.12.4 Layer 2 Repeater Address Encoding**:

> The repeater address subfield is encoded in the same manner as the destination and source  address subfields, except for the most-significant bit in the last octet, called the H bit. As discussed in Section 3.12.1, the H bit indicates whether a frame has been repeated or not. 

In the TNC2 packet monitoring format used on the APRS-IS, the last digipeater callsign having the H bit set is marked with a '*'. For example,

`SRC>DST,VIA1,VIA2*,VIA3,VIA4:data`

Or, live:

`K2PUT-15>APRS,NA1RA-12,WIDE2*,qAR,KC1SPC-15:!4127.67NP07343.51W#PHG2250/PEARL 145.130-PL136.5`

In this case NA1RA-12 and WIDE2 both had the H bit set, but the asterisk is only appended on the last VIA field (WIDE2).

mxi-box, 2024-03-30:
I‘m currently working on some feature of AX25, maybe I can fix this btw.

Is the reference i reading now [https://www.tapr.org/pdf/APRS101.pdf](url) correct? 

I saw some TNC2 messages are constructed with many asterisks, such as:

`BH6BEZ-1>APRS,BH5HQJ-8*,WIDE1*,WIDE2-1,qAR,BA5AG-10:!3011.73N/12011.24E-PHG5530 SDRangel 144.390MHz/144.640MHz`

```BG5AJN-8>RYUXV8,BG5AJN-1*,WIDE1*,BR5AS-10*,WIDE2*,qAS,BD5DPP-10:`0<jm!U>/`"46}shaoxing club_"```

Are these above both kinda _nonstandard_ messages?

closed 2024-03-30 by f4exb

hessu, 2024-04-05:
Unfortunately there is no proper standard for the TNC2 monitor format. Some TNCs apparently used asterisks on all digipeater calls which had the H bit on, some used an asterisk only on the last one.

PR #2039 (2024-03-30): AX.25 Packet: packetmod supports multi Via; packetdemod fix H bit
Can use multi VIAs like `VIA1,VIA2,VIA3` or no VIA at all

Fix #2037