f4exb/sdrangel #2047: Enter centerfreq don't work on android

DickVos, 2024-04-02:
Hello,

Does anybody have the same issue that you can not enter the centre freq or BW? 

This is on an android device. The keyboard appears when you press a digit but whatever you type I don't change the digit?

Dick

srcejon, 2024-04-03:
Similar issue reported here: https://github.com/f4exb/sdrangel/issues/1732

What version of Android and device?

Unfortunately, I can't reproduce it, so don't know how to fix it.

DickVos, 2024-04-04:
Thanks for respond,

I have the Galaxy Tab II with Android 9

abelouso, 2024-04-08:
I have the same problem, Galaxy Tab A 8.0, Android 9

DickVos, 2024-04-09:
Looks like an Android 9 I.c.w. SDR Angel issue

DickVos, 2024-04-22:
I have installed GBoard, Google keyboard from the playstore and set it as default. Now it's working. Problem solved.

closed 2024-06-07 by github-actions[bot]