f4exb/sdrangel #2050: Hermes Lite 2 Full Support

CRD716, 2024-04-05:
The [Hermes Lite 2](http://www.hermeslite.com/) is a pretty common open source SDR, and I think many would enjoy it being fully supported in sdrangel. I know it works somewhat as a MIMO device, but last I saw it was missing some features like N2ADR filter board control. I could also be completely blind/making this up, but this is what I've read/found in the documentation so far.

https://github.com/softerhardware/Hermes-Lite2/wiki/Software has a list of other programs that currently support it, along with some other info.

CRD716, 2024-05-20:
_Cough_

CRD716, 2024-07-01:
cough cough cough

CRD716, 2024-08-10:
but what if

Cisien, 2024-08-21:
This would be amazing. I just picked up a Hermes Lite 2, and find the current supported linux SDR options a bit clunky to use.

SJRDorset, 2024-09-08:
I completely agree. It would be really good to have complete support for the Hermes Lite 2. I also have an Anan 10 and it would be nice if this was supported as well.

closed 2024-10-24 by github-actions[bot]