f4exb/sdrangel #2053: I can't compile it on Kali Linux

Electro-Gamma, 2024-04-05:
┌──(root㉿kali)-[~]
└─# cd /opt/build
#git clone https://github.com/f4exb/sdrangel.git
cd sdrangel
mkdir build; cd build   
cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DMIRISDR_DIR=/opt/install/libmirisdr \
-DAIRSPY_DIR=/opt/install/libairspy \
-DAIRSPYHF_DIR=/opt/install/libairspyhf \
-DBLADERF_DIR=/opt/install/libbladeRF \
-DHACKRF_DIR=/opt/install/libhackrf \
-DRTLSDR_DIR=/opt/install/librtlsdr \
-DLIMESUITE_DIR=/opt/install/LimeSuite \
-DIIO_DIR=/opt/install/libiio \
-DPERSEUS_DIR=/opt/install/libperseus \
-DXTRX_DIR=/opt/install/xtrx-images \
-DSOAPYSDR_DIR=/opt/install/SoapySDR \
-DUHD_DIR=/opt/install/uhd \
-DAPT_DIR=/opt/install/aptdec \
-DCM256CC_DIR=/opt/install/cm256cc \
-DDSDCC_DIR=/opt/install/dsdcc \
-DSERIALDV_DIR=/opt/install/serialdv \
-DMBE_DIR=/opt/install/mbelib \
-DCODEC2_DIR=/opt/install/codec2 \
-DSGP4_DIR=/opt/install/sgp4 \
-DLIBSIGMF_DIR=/opt/install/libsigmf \
-DDAB_DIR=/opt/install/libdab \
-DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
make -j $(nproc) install
mkdir: cannot create directory ‘build’: File exists
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.43.0") 
-- SDRangel Version: 7.19.1
-- Build type set to: RelWithDebInfo
-- Compiling for 24 bit Rx DSP chain
-- Looking for __x86_64__
-- Looking for __x86_64__ - found
-- Target architecture: x86_64-native
-- Performing Test FLAG_SUPPORTED
-- Performing Test FLAG_SUPPORTED - Success
-- Looking for __ARM_NEON
-- Looking for __ARM_NEON - not found
-- Looking for __AVX512F__
-- Looking for __AVX512F__ - not found
-- Looking for __AVX2__
-- Looking for __AVX2__ - not found
-- Looking for __AVX__
-- Looking for __AVX__ - found
-- Looking for __SSE4_2__ - forced found
-- Looking for __SSE4_1__ - forced found
-- Looking for __SSSE3__ - forced found
-- Looking for __SSE3__ - forced found
-- Looking for __SSE2__ - forced found
-- Interprocedural optimization enabled
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.83.0/BoostConfig.cmake (found version "1.83.0")  
-- Checking for module 'fftw3f >= 3.0'
[... 146 lines cut ...]
  add_subdirectory given source "plugins" which is not an existing directory.                           
                                                                                                        
                                                                                                        
CMake Error at CMakeLists.txt:866 (add_subdirectory):
  add_subdirectory given source "sdrsrv" which is not an existing directory.                            
                                                                                                        
                                                                                                        
CMake Error at CMakeLists.txt:869 (add_subdirectory):
  add_subdirectory given source "plugins" which is not an existing directory.                           
                                                                                                        
                                                                                                        
-- Found Doxygen: /usr/bin/doxygen (found version "1.9.8") found components: doxygen dot 
-- added optional 'make dev_docs' for developer documentation generation
-- Found Dot: /usr/bin/dot  
-- Check for case-sensitive file systems
-- File system is case-sensitive
-- Asking qmake for QT_PLUGINS_DIR and got /usr/lib/x86_64-linux-gnu/qt5/plugins
-- Asking qmake for QT_IMPORTS_DIR and got /usr/lib/x86_64-linux-gnu/qt5/imports
-- Asking qmake for QT_QML_DIR and got /usr/lib/x86_64-linux-gnu/qt5/qml
-- Asking qmake for QT_LIBS_DIR and got /usr/lib/x86_64-linux-gnu
-- CPack generators: 
-- Configuring incomplete, errors occurred!
make: *** No rule to make target 'install'.  Stop.

> ``

srcejon, 2024-04-05:
Did the git clone work? Do the reported missing directories exist?

Electro-Gamma, 2024-04-05:
> Did the git clone work? Do the reported missing directories exist?

yes of course everything here :
https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

Electro-Gamma, 2024-04-05:
all cloned git in /opt/build/

srcejon, 2024-04-05:
Two things:

    mkdir: cannot create directory ‘build’: File exists

Perhaps delete build and start from scratch.

Also, what are all the `virtual:world` messages?

Electro-Gamma, 2024-04-06:
> Two things:
> 
> ```
> mkdir: cannot create directory ‘build’: File exists
> ```
> 
> Perhaps delete build and start from scratch.
> 
> Also, what are all the `virtual:world` messages?

I don't know, everything is correct but same problem 
on kali linux 2024.1
arm64 and amd64

closed 2024-05-21 by github-actions[bot]