f4exb/sdrangel #2060: ChannelAPI::getIndexInDeviceSet() not valid in channelAdded slot

srcejon, 2024-04-08:
In a MainCore::channelAdded(int deviceSetIndex, ChannelAPI *channel) slot, it appears channel->getIndexInDeviceSet() is not valid.

This seems to come from the ordering of the following code:

    void DeviceSet::addChannelInstance(ChannelAPI *channelAPI)
       ...
        mainCore->addChannelInstance(this, channelAPI);
        renameChannelInstances();

addChannelInstance emits the channelAdded signal, but the index isn't set until afterwards in renameChannelInstances().

f4exb, 2024-04-09:
Inverting the two lines may just work:
```c++
void DeviceSet::addChannelInstance(ChannelAPI *channelAPI)
{
    MainCore *mainCore = MainCore::instance();
    m_channelInstanceRegistrations.append(channelAPI);
    renameChannelInstances();
    mainCore->addChannelInstance(this, channelAPI);
}
```
The "append" adds the channel at the end if the `m_channelInstanceRegistrations` list then `renameChannelInstances()` will just loop through this list and set the channel index in device set to the current index. This sets this index in the `channelAPI` to the last index which makes sense. Before `renameChannelInstances()` is invoked this index may not be set correctly.

In seems to me a similar situation exists in other parts of the code:
  - `addMIMOchannel`
  - `addRxChennel`
  - `addTxChannel`

One may also take a look at the channel removal methods although I suppose that what we want here is the index of the channel that has been deleted which in this case is correct.

srcejon, 2024-04-09:
Swapping the order seemed to work, for what I was looking at, but renameChannelInstances can emit an indexInDeviceSetChanged signal - and I don't know if there's any code that might be dependent on the ordering of those two signals. Couldn't see anything from a quick search.
.

closed 2024-05-26 by github-actions[bot]