f4exb/sdrangel #2061: Qt6 - multiple device sets do not load correctly

srcejon, 2024-04-08:
When built with Qt6, if the default configuration has multiple device sets, one of the devices may not be correctly restored, when SDRangel is started (On Windows and Mac, haven't tried Linux yet).

- Create a workspace with two Test Sources
- Restart SDRangel
- Sometimes it creates one Test Source and one File Input, occasionally just one Test Source.

srcejon, 2024-04-08:
2024-04-08 17:32:15.410 (D) MainWindow::MainWindow: load current configuration...
2024-04-08 17:32:15.411 (D) MainWindow::loadConfiguration: configuration [default | no name] 1 workspace(s) - 2 device set(s) - 0 feature(s)
2024-04-08 17:32:15.416 (D) *MainWindow::loadConfiguration: Unknown preset type:  7602297*
2024-04-08 17:32:15.416 (D) DeviceEnumerator::getBestSamplingDeviceIndex: serial matched (exact): id: sdrangel.samplesource.testsource ser: 
2024-04-08 17:32:15.417 (D) MainWindow::loadConfiguration: add source sdrangel.samplesource.testsource in workspace 0 spectrum in 0

Unknown preset type is perhaps the problem.

srcejon, 2024-04-08:
The problem is in the Preset copy constructor:

    Preset::Preset(const Preset& other) :
	    m_group(other.m_group),
   	    m_description(other.m_description),
  	    m_centerFrequency(other.m_centerFrequency),
 	    m_spectrumConfig(other.m_spectrumConfig),
	    m_dcOffsetCorrection(other.m_dcOffsetCorrection),
	    m_iqImbalanceCorrection(other.m_iqImbalanceCorrection),
	    m_channelConfigs(other.m_channelConfigs),
  	    m_deviceConfigs(other.m_deviceConfigs),
 	    m_showSpectrum(other.m_showSpectrum),
	    m_layout(other.m_layout)

It doesn't copy all of the members. It's missing some of:

    m_presetType(other.m_presetType),
    m_spectrumGeometry(other.m_spectrumGeometry),
    m_spectrumWorkspaceIndex(other.m_spectrumWorkspaceIndex),
    m_deviceGeometry(other.m_deviceGeometry),
    m_deviceWorkspaceIndex(other.m_deviceWorkspaceIndex),
    m_selectedDevice(other.m_selectedDevice),

Are some of them not meant to be copied? m_presetType and m_selectedDevice are what's causing this problem, I think.

Also, the resetToDefaults method doesn't reset everything.

f4exb, 2024-04-08:
It does not seem to me there is a specific reason behind these missing parts.

closed 2024-04-12 by srcejon