f4exb/sdrangel #2064: Fix wiki page Compile-from-source-in-Linux.md

dforsi, 2024-04-13:
The attached patch fixes two small issues in the documentation.

If you did copy and paste the command suggested in the terminal you would get errors and fail to install all needed packages because some lines containing the names of the packages to be installed would be interpreted by the shell as commands:
```
E: Unable to locate package  
-bash: qml6-module-qtquick-dialogs: command not found
-bash: qt6-webengine-dev-tools: command not found
```
[0001-Fix-continuation-lines.patch.txt](https://github.com/f4exb/sdrangel/files/14966933/0001-Fix-continuation-lines.patch.txt)

dforsi, 2024-04-13:
The following packages are missing from the list of dependencies for Qt6 in `Compile-from-source-in-Linux.md`:
- qt6-svg-dev
- qt6-location

f4exb, 2024-04-16:
Thanks, done.

closed 2024-04-16 by f4exb