f4exb/sdrangel #2065: Device settings changes are not communicated via Reverse API

godsic, 2024-04-13:
When Reverse API is activated for device, the corresponding server receives all device settings via PATCH method as expected. Any subsequent device settings changes, notably center frequency, are not communicated to the server neither through PATCH or PUT requests.

srcejon, 2024-04-13:
Which device?

godsic, 2024-04-13:
@srcejon rtlsdr

godsic, 2024-04-14:
@srcejon According to this logic:

```cpp
 if (settingsKeys.contains("useReverseAPI"))
    {
        bool fullUpdate = (settingsKeys.contains("useReverseAPI") && settings.m_useReverseAPI) ||
            settingsKeys.contains("reverseAPIAddress") ||
            settingsKeys.contains("reverseAPIPort") ||
            settingsKeys.contains("reverseAPIDeviceIndex");
        webapiReverseSendSettings(settingsKeys, settings, fullUpdate || force);
    }
```

The communication would only happen if `settingsKeys.contains("useReverseAPI")`, i.e., if `useReverseAPI` is changed. Shouldn't this be something like ` if (settingsKeys.contains("useReverseAPI") || settings.m_useReverseAPI)`?

PR #2066 (2024-04-14): Fix bug that prevents settings changes updates via reverse API (#2065)
Most plugins that use reverse API to PATCH settings updates to remote server only do so when `useReverseAPI` is toggled, but not when the relevant settings are being updated. So lets fix the precondition to use the `m_useReverseAPI` flag instead.

Fixes #2065.
closed 2024-04-14 by f4exb