f4exb/sdrangel #2071: Implemented in 7.20  Map feature can export data to KML files.

choupette-change, 2024-04-16:
Implemented in 7.20 
Map feature can export data to KML files. Implements AIS and ADSB decoders: improvement proposal #1068
Enhancement request, can you provide a real-time KML feed to power a related application ?

srcejon, 2024-04-17:
KML export was implemented via Cesium, which doesn't support real-time export.

This is similar to #2012

alphafox02, 2024-04-20:
I figured there’s something telling cesium where to plot, couldn’t those same functions be where “CoT” is optionally introduced in order to populate an external map or rather a TAK server which would in turn populate numerous maps depending on how the TAK server is configured.

Here is an example in C++
https://github.com/chipbrommer/COT_Utility

srcejon, 2024-04-20:
Either can be done, but I have a long list of other stuff to do first, I'm afraid.

alphafox02, 2024-04-21:
Totally understand. Awesome work you all are doing.

closed 2024-06-05 by github-actions[bot]