f4exb/sdrangel #2077: qt6-location not fount in repository

jlrector2000, 2024-04-20:
Attempting to build for Qt6 for Ubuntu 24.04, per build instructions.
All dependencies downloaded with 'apt-get install' except for "qt5-location",
which errored with 'file not found'.
A web search also could not locate this file.  What to do???

srcejon, 2024-04-20:
24.04 uses qt 6.4, which does not have location. You need qt 6.5 for it.

jlrector2000, 2024-04-20:
Sorry for the typo above...meant to type "qt6-location" which errored out.
For us nubies, would you further explain "24.04 uses qt 6.4, which does not have location. You need qt 6.5 for it."
Thanx!

srcejon, 2024-04-20:
> . For us nubies, would you further explain "24.04 uses qt 6.4, which does not have location. You need qt 6.5 for it." Thanx!

Ubuntu 24.04 uses Qt version 6.4, and this does not have the Qt Location package. It was only added to Qt 6 in Qt 6.5.

I'd recommend you just build with Qt 5 (as that has everything you need and there's no advantage to using Qt 6 anyway).

jlrector2000, 2024-04-21:
I did that, and it works perfectly...just was trying to go upwards to Qt 6.  Thanx!

f4exb, 2024-04-21:
Seems we're not ready yet to mass migrate to Qt6. Just making sure the code is compatible.

closed 2024-04-21 by f4exb