f4exb/sdrangel #2081: SDRangel does not start on Fedora Silverblue

GeraldSoellinger, 2024-04-23:
I installed sdrangel via "Software" app and then tried to start it via the sdrangel icon.
The system tried to load it for some seconds, but then quit.

Any idea what the problem is? 

Below my system config:

SDRangel Version 7.20.0 from dl.flathub.org

## Software Information:
- **Firmware-Version:**                            18ee55c
- **Name des Betriebssystems:**                    Fedora Linux 39.20240414.0 (Silverblue)
- **Betriebssystem-Build:**                        (null)
- **Betriebssystem-Typ:**                          64-bit
- **GNOME-Version:**                               Nicht verfügbar
- **Fenstermanager:**                              Wayland
- **Kernel-Version:**                              Linux 6.8.5-201.fc39.x86_64

GeraldSoellinger, 2024-04-23:
sudo dmesg shows following output:
 traps: sdrangel[7131] trap invalid opcode ip:7f89cb56e384 sp:7ffef6ff8fe0 error:0 in libsdrbase.so[7f89cb343000+242000]

srcejon, 2024-04-23:
I would guess your CPU doesn't support SSE 4.2

GeraldSoellinger, 2024-04-23:
There is a flag with sse4_2, I would guess it is supported.
 lscpu
Architecture:             x86_64
  CPU op-mode(s):         32-bit, 64-bit
  Address sizes:          48 bits physical, 48 bits virtual
  Byte Order:             Little Endian
CPU(s):                   8
  On-line CPU(s) list:    0-7
Vendor ID:                AuthenticAMD
  Model name:             AMD Opteron(tm) Processor 4256 EE
    CPU family:           21
    Model:                1
    Thread(s) per core:   2
    Core(s) per socket:   4
    Socket(s):            1
    Stepping:             2
    Frequency boost:      enabled
    CPU(s) scaling MHz:   64%
    CPU max MHz:          1600,0000
    CPU min MHz:          1000,0000
    BogoMIPS:             3199,96
    Flags:                fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge m
                          ca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall
                           nx mmxext fxsr_opt pdpe1gb rdtscp lm constant_tsc rep
                          _good nopl nonstop_tsc cpuid extd_apicid aperfmperf pn
                          i pclmulqdq monitor ssse3 cx16 sse4_1 sse4_2 popcnt ae
                          s xsave avx lahf_lm cmp_legacy svm extapic cr8_legacy 
                          abm sse4a misalignsse 3dnowprefetch osvw ibs xop skini
                          t wdt lwp fma4 topoext perfctr_core perfctr_nb cpb hw_
                          pstate ssbd vmmcall arat npt lbrv svm_lock nrip_save t
                          sc_scale vmcb_clean flushbyasid decodeassists pausefil
                          ter pfthreshold
Virtualization features:  
  Virtualization:         AMD-V
Caches (sum of all):      
  L1d:                    128 KiB (8 instances)
  L1i:                    256 KiB (4 instances)
  L2:                     8 MiB (4 instances)
  L3:                     8 MiB (1 instance)
NUMA:                     
  NUMA node(s):           1
  NUMA node0 CPU(s):      0-7
Vulnerabilities:          
  Gather data sampling:   Not affected
  Itlb multihit:          Not affected
  L1tf:                   Not affected
  Mds:                    Not affected
  Meltdown:               Not affected
  Mmio stale data:        Not affected
  Reg file data sampling: Not affected
  Retbleed:               Mitigation; untrained return thunk; SMT vulnerable
  Spec rstack overflow:   Not affected
  Spec store bypass:      Mitigation; Speculative Store Bypass disabled via prct
                          l
  Spectre v1:             Mitigation; usercopy/swapgs barriers and __user pointe
                          r sanitization
  Spectre v2:             Mitigation; Retpolines; STIBP disabled; RSB filling; P
                          BRSB-eIBRS Not affected; BHI Not affected
  Srbds:                  Not affected
  Tsx async abort:        Not affected

GeraldSoellinger, 2024-04-23:
here some further info to my hardware
# Systemdetailsbericht
---

## Berichtdetails
- **Erstellungsdatum:**                            2024-04-23 09:57:12

## Hardware Information:
- **Hardware-Modell:**                             ASUS KCMA-D8
- **Speicher:**                                    15,5 GiB
- **Prozessor:**                                   AMD Opteron™ 4256 EE × 8
- **Grafik:**                                      NV106
- **Festplattenkapazität:**                        250,1 GB

GeraldSoellinger, 2024-04-23:
And I use following receiver:
[  520.562356] usb 2-1: Product: NESDR SMArt v5
[  520.562363] usb 2-1: Manufacturer: Nooelec

GeraldSoellinger, 2024-04-23:
seems like a Pulsaudio config is missing. Why is a pulseaudio config needed and what to configure?

:~/Downloads$ flatpak -v run org.sdrangel.SDRangel 
F: No installations directory in /etc/flatpak/installations.d. Skipping
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /var/home/gerald/.local/share/flatpak
F: Opening user flatpak installation at path /var/home/gerald/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: Opening user flatpak installation at path /var/home/gerald/.local/share/flatpak
F: Opening system flatpak installation at path /var/lib/flatpak
F: /var/lib/flatpak/runtime/org.kde.Platform/x86_64/5.15-23.08/ad419c00787e55313706a937198d82aa5050d94db33ea7e229a16dc780cd4889/files/lib32 does not exist
F: Cleaning up unused container id 13512353
F: Cleaning up per-app-ID state for org.sdrangel.SDRangel
F: Allocated instance id 651754608
F: Add defaults in dir /org/sdrangel/SDRangel/
F: Add locks in dir /org/sdrangel/SDRangel/
F: Allowing x11 access
F: Allowing pulseaudio access
F: Pulseaudio user configuration file '/var/home/gerald/.config/pulse/client.conf': Fehler beim Öffnen der Datei »/var/home/gerald/.config/pulse/client.conf«: Datei oder Verzeichnis nicht gefunden
F: Running '/usr/bin/bwrap --args 47 /usr/bin/xdg-dbus-proxy --args=49'
F: Running '/usr/bin/bwrap --args 47 sdrangel'

srcejon, 2024-04-23:
Looks like there may be AVX2 instructions in the flatpak (vpmaskmovd), which your CPU doesn't support.

@AsciiWolf - Looking at: https://github.com/flathub/org.sdrangel.SDRangel/blob/master/org.sdrangel.SDRangel.yaml the sdrangel rule doesn't seem to use -DARCH_OPT= option to compile for older architectures. Perhaps it should?

AsciiWolf, 2024-04-23:
Thanks for pointing out the AVX2 instruction set! I wonder, are there any possible performance drawbacks on newer CPUs if I set the `ARCH_OPT` option to use SSE4_2.

AsciiWolf, 2024-04-23:
Also, please, if possible, move this ticket [here](https://github.com/flathub/org.sdrangel.SDRangel/issues). Thanks!

srcejon, 2024-04-23:
> Thanks for pointing out the AVX2 instruction set! I wonder whether there are any possible performance drawbacks on newer CPUs if I set the `ARCH_OPT` option to use SSE4_2.

Possibly a small hit, but I would think it be better that the flatpak (and snap, that I've just fixed as well), run on as many machines as possible (within reason). If someone wants an optimized version, they can build from source.

It's a bit tricky supporting different ISA extensions within a single binary (although can be done).

closed 2024-06-09 by github-actions[bot]