f4exb/sdrangel #2082: Data passthrough plugin from one device (RX) to another one (TX)

Blukain, 2024-04-23:
Hello everyone
I'm new to SDRAngel, was digging a bit in plugin development and couldn't find a possible way to implement what I need, wondering if the software lack a pipe from rx to tx.
The goal is to receive a signal from device, use a single plugin rx/tx (or two, 1rx +1tx) to manage data and modify/clean it, for after be able to send new data to another device (not same as receiving one, not mimo) connected and transmit cleaned data.
I really appreciate any advice on how to achieve this if possible... or get to know if is not due to lack of support (as common rx/tx pipe).

Thanks you so much!

f4exb, 2024-04-24:
It depends on what you mean by "modify/clean it". There are indirect ways to do this. 

For example to implement a repeater use case you can use the system audio to do the plumbing. Implementing the controls is another story but for what concerns the audio stream it should be possible to implement the use case.

You have the UDP source and sinks that can be tied together using the loopback interface. You could pipe I/Q streams like this (or demodulated signals). By using the loopback network interface it is just as if using a FIFO but managed by the network layer.

Finally you have the local source and sinks but it cannot be used as a source in a Tx stream. It may be possible to link it to a local source just as we do with the UDP. There is a catch however since Rx and Tx clocks may have some skew. It has decimation and FFT filtering capabilities

srcejon, 2024-04-24:
@f4exb - he's writing his own channel plugins (see [url=https://sdrangel.org/forum/viewtopic.php?t=170&start=10]here[/url), so I would have though Data Pipes would probably be the way to go?

f4exb, 2024-04-25:
OK I was not aware of the discussion. So basically this leads us to my third point. For now there is a segregation between Rx and Tx I/Q flows. To be able to import I/Q from a Rx channel in a Tx chain you need a Tx channel intermediate. The Local Source plugin could be this intermediate if we augment it with the capability of taking the I/Q samples from a Rx channel capable of producing I/Q samples. The component to actually perform the plumbing can indeed be a data pipe.

At some point we have to deal with Rx / Tx clock skew. Unless both run on the same clock which is usually not the case you will have to manage the respective clocks drift. This can be implemented in the Local Source plugin. For now as we take the samples internally from a Local Output plugin the whole chain is driven by the Tx device clock that orchestrates the pulling of samples. This will not be the case if these samples originally come from a random Rx device.

Blukain, 2024-04-25:
Thanks you so much for your information
So basically there is segregation between rx/tx channels, I have already built to plugins a rx and one tx, for what i understood it might be possible to connect them but need to add the pipe that takes I/Q from the rx correct?
Which might be the interface to use, through the baseband or channelAPI?
What about feature plugin, is only for input (rx channel plugin) correct? it only gets I/Q samples there is no way to use it as a bridge to a tx channel plugin?

Thanks you so much @f4exb and @srcejon

f4exb, 2024-04-25:
Feature plugins are there to support functionalities not directly related to an I/Q stream. The Demod Analyzer does consume I/Q samples but is not limited to complex I/Q and is not directly linked to the I/Q stream produced by the Rx device. It takes samples from a channel plugin that has already processed the I/Q samples of the device in some way and it can be used as a virtual oscilloscope.

Here you will need a Tx channel plugin to re-introduce your I/Q samples in a Tx chain. The closest I can think of is the Local Source. At the moment it is limited to taking samples from a Local Output device but it could be extended to take its samples to some components (plugins) that produce I/Q samples hence your Rx channel plugin in this use case.

Blukain, 2024-04-25:
Hello @f4exb
I am stuck on this part "Here you will need a Tx channel plugin to re-introduce your I/Q samples in a Tx chain", what do you mean to re-introduce? baseband or other?
I took a look at local source plugin but got lost on trying to make a pipe connection with a specific plugin or channel made by my rx plugin.....

Any suggestion?

Thanks you so much, appreciated

f4exb, 2024-04-26:
> I thought that should be the rx plugin to send data back again.

No... this is not very clean regarding the whole design. You have to go through the intermediate of a channel plugin. Moreover if you use the Local Source it allows further Interpolation if necessary and some shifting inside the baseband.

> I took a look at local source plugin but got lost on trying to make a pipe connection with a specific plugin or channel made by my rx plugin.....

I am sure some examples exist. I think the Demod Analyzer feature plugin connects to Rx channels to obtain their demodulated or processed samples. This should not be much different.

Blukain, 2024-04-26:
Hello @f4exb

Thanks you so much im making modification on a local source copy.
But i would like if i can somehow debug the code, it keeps crashing when i try to set the RxChannel...
I used the function void DemodAnalyzer::setChannel(ChannelAPI *selectedChannel) copied from demodAnalyzer...
I am currently developing plugins in qtcreator with qt6.5.3 opensource

Thanks you so much

f4exb, 2024-04-26:
Hello @Blukain It will be very difficult to help you if I can't see the code. Have you forked this repo and committed your changes?

closed 2024-06-11 by github-actions[bot]