f4exb/sdrangel #2083: crashes during long heatmapping drives(over 30 min)

Sparkycivic, 2024-04-24:
Per my Reddit thread, 
I've been struggling with app crashes while drive mapping, takes a little over half an hour to occur. "Keep screen on" is ticked, and it doesn't seem to matter if i'm interacting with the app or not, it just disappears and all map data lost. I have developed a 'habit' of stopping to save CSV, then continue mapping, which works ok. 

maybe related: SDRAngel must be in foreground or else it stops collecting/crashes after a very short time... is that part expected, or is it a clue?

Galaxy S23 with 8GB RAM, 
SM-S911W with stock ROM.
Android 14, OneUI 6.1
RTL_SDR generic dongle with Q-branch port/filter"full band" option. 
Doesn't seem to matter if I'm mapping in Q branch or tuner. 
External power(Dock/hub) or standalone both tested.

srcejon, 2024-04-24:
From the dicussion on Reddit, I suspect this may be an out of memory issue. As mapping / charts aren't needed, I'll add an option to disable these, as it should reduce memory requirements significantly, and also add some extra debug.

Sparkycivic, 2024-04-24:
Thank you.

I also plan to try some of the other suggestions from the Reddit thread regarding power , and using my alternate phone device. An opportunity for a session to test for that should come in the following days.

Can I ask, is it possible to help with this potential memory issues by having an option to change the data interval from "x records per second, to "1 record per x meters"? That would make it work exactly as my huge Anritsu SiteMaster in mapping mode, and can really save on data volumes.

⁣Get TypeApp for Android ​

Sparkycivic, 2024-08-14:
I regret to report that the crashing continues. I recently went for a couple hours' drive and found that the app seems to either freeze, or crash after around 15-25 minutes of steady-state data collecting. It doesn't seem to matter if I save the .CSV file after 15 minutes(it's super hard to remember to pull over and fiddle with the phone when I'm by myself) and start anew, it will still crash eventually. I wasn't able to identify where/how to alter the data collection settings to reduce the number of points, nor how to disable the map generation aspect of the app which may or may not be contributing to the amount of memory being taken. 
I use the app with my Galaxy S23 SM-S911W, with Android 14, qualcomm cpu and 7 gigs of ram according to cpu-z app. I usually use it with portrait-orientation fixed, and "keep screen on" set.

srcejon, 2024-08-15:
To minimize memory usage:

- Uncheck the checkboxes next to Record for data you don't need - E.g. Max,Min,Pulse, Path Loss etc.
- Uncheck the Display Chart button, so the chart isn't displayed

You can set Data to None, which will stop the image from being created - but I've just realised this needs to be set in order to save .csv data so isn't going to help you. I'll need to add a separate button.

Sparkycivic, 2024-08-26:
OK, had a chance to go for another drive.  I now believe that the the culprit was the display chart. It has not crashed once since keeping the chart display off, or clearing the chart regularly by toggling it on/off while keeping the heatmap csv data. I'm guessing that with the chart left to run, it overflows with data after the 20 min period(longest setting), thus crashing the app. Perhaps it can be fixed to side-scroll the chart, and discard the older data? I like having the long-term window display because I'm measuring broadcast FM and other wide-area signals, so being able to see in realtime, how measurements are trending vs time is really insightful. I can use the graph and my visual situation to learn about the RF environment.  I have been able to drive for an hour, and save all of the data to one single CSV at the end. 
Feature request: Is it possible to save/amend to CSV continuously (at some regular interval maybe 1 minute)? 
Feature request: "Keep display on" setting save alongside the other settings which are captured with the "save all settings" command?  I find some difficulty to remember to put the display setting each launch, which pauses gathering data as soon as the phone goes to lock.

srcejon, 2024-09-25:
> I now believe that the the culprit was the display chart. 

Yep - in HeatMapGUI::addToPowerSeries we're not catching std::bad_alloc if we run out of memory while adding data to a chart series.