f4exb/sdrangel #2084: Sdrangel on Intel(R) Core(TM) i5-3230M CPU with flag sse4_2 in DEBIAN 12 does not start.

Lat6819, 2024-04-24:
Hi, I have old notebook X230 with Intel(R) Core(TM) i5-3230M CPU. I have checked twice - my CPU has SSE 4.2. I checked on internet and with command lscpu. So my CPU is capable to run SDRANGEL.
I have tried FLATHUB and SNAP. I have tried X11 and WAYLAND. SDRANGEL does not start. 
System: 6.1.0-20-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.85-1 (2024-04-11) x86_64 GNU/Linux - Debian 12
Flatpak 1.14.4
snap    2.61.2
snapd   2.61.2
series  16
debian  12
kernel  6.1.0-20-amd64

srcejon, 2024-04-24:
The flatpak and snap both require AVX2, not just SSE 4.2 (although we're currently in the process of changing this to just SSE 4.2)

srcejon, 2024-04-25:
A new snap has been uploaded, so you can give that a try to see if it's any better.

closed 2024-06-10 by github-actions[bot]