f4exb/sdrangel #2085: Cannot build with vkFFT enabled on Linux

agatti, 2024-04-25:
When building current git HEAD with the "-DVKFFT_BACKEND=0" flag, 
the build script doesn't find HLSL support in the latest Vulkan SDK for Linux (1.3.280.1 at the time of writing): 

```bash
$ source ~/.local/vulkan/setup-env.sh
$ cmake -B build -S ~/src/sdrangel -DVKFFT_BACKEND=0
# ...
-- Found WrapVulkanHeaders: /home/agatti/.local/vulkan/x86_64/include
# ...
-- Found Vulkan: /home/agatti/.local/vulkan/x86_64/lib/libvulkan.so (found version "1.3.280") found components: glslc glslangValidator
CMake Error at sdrbase/CMakeLists.txt:609 (find_library):
  Could not find VULKAN_HLSL_LIB using the following names: HLSL
```

My CMake's (3.29.1) `FindVulkan.cmake` file doesn't mention `HLSL` among the components it can look for.  I'm not a Vulkan expert, but if it really needs to compile HLSL shaders into SPIR-V then shouldn't it look for `dxc` (the DirectX shader compiler) instead?

srcejon, 2024-04-25:
Don't know - but looking back at the pull request https://github.com/f4exb/sdrangel/pull/1779 it worked with vulkansdk-linux-x86_64-1.3.250.1.tar.gz and cmake 3.22.1

To be honest, there's probably not much point using it at the moment anyway, as it's usually slower than FFTW.

agatti, 2024-04-25:
I didn't know vkFFT was slower at the moment.  On the other hand I'm using a rather slow machine to begin with so maybe that can help still.

That said, with 1.3.250.1 CMake works as expected.  I've taken a quick look at what's different between those two SDKs and from what I can see later Vulkan SDKs handle HLSL support differently:

(1.3.250.1):

```$ fd hlsl                                                                               
x86_64/include/glslang/HLSL/
x86_64/include/glslang/HLSL/hlslAttributes.h
x86_64/include/glslang/HLSL/hlslGrammar.h
x86_64/include/glslang/HLSL/hlslOpMap.h
x86_64/include/glslang/HLSL/hlslParseHelper.h
x86_64/include/glslang/HLSL/hlslParseables.h
x86_64/include/glslang/HLSL/hlslScanContext.h
x86_64/include/glslang/HLSL/hlslTokenStream.h
x86_64/include/glslang/HLSL/hlslTokens.h
x86_64/include/spirv_cross/spirv_hlsl.hpp
x86_64/lib/libHLSL.a
x86_64/lib/libHLSLTestLib.a
x86_64/lib/libLLVMHLSL.a
x86_64/lib/libspirv-cross-hlsl.a
```

(1.3.280.1):

```$ fd hlsl
x86_64/include/spirv_cross/spirv_hlsl.hpp
x86_64/lib/libHLSLTestLib.a
x86_64/lib/libLLVMHLSL.a
x86_64/lib/libspirv-cross-hlsl.a
x86_64/share/spirv_cross_hlsl/
x86_64/share/spirv_cross_hlsl/cmake/spirv_cross_hlslConfig-release.cmake
x86_64/share/spirv_cross_hlsl/cmake/spirv_cross_hlslConfig.cmake
```

I bet those extra `.cmake` files need to be taken into account for recent SDKs.

agatti, 2024-04-25:
Yep, changed in [1.3.275.0](https://vulkan.lunarg.com/doc/sdk/1.3.275.0/linux/release_notes.html):

> The glslang legacy libraries named HLSL and OGLCompiler have been removed. To avoid future disruptions, please use cmake's find_package mechanism rather than hard coding library dependencies. Only the headers that are part of glslang's public interface are included in the install target.

agatti, 2024-04-26:
I guess this can now be closed as the build fix was merged.

closed 2024-04-26 by agatti