f4exb/sdrangel #2090: Display problem with the latest versions of Ubuntu.

choupette-change, 2024-04-26:
For your information, with the latest versions of Ubuntu 23 10 and 24 04 LTS, it works except for the "Waveform" item. The screenshot does not represent the real problem, it would be a problem updating this part, all the curves accumulate and end up making a big spot of color..
[image]

srcejon, 2024-04-28:
In 24.04 under VM Workstation, I get:

[image]

choupette-change, 2024-04-28:
I have the same problem as you in virtual machine VM 24 04 and installation Ubuntu 24 04 dual boot with Windows.

f4exb, 2024-05-16:
it appears that 24.04 is not available yet using the do release upgrade method. They say that there are some issues and they wait for 24.04.1 to be released in August

f4exb, 2024-05-24:
I have compiled SDRangel on my brand new RasPi 5 on Ubuntu 24.04 and did not notice anything wrong with the display.

closed 2024-07-09 by github-actions[bot]