f4exb/sdrangel #2094: AIS Message #14 - message string truncation

madmanguruman, 2024-04-29:
For AIS message 14, the data display is not showing the message correctly.

Example: An AIS message 14 is encoded with the message **MOB TEST** and transmitted. SDRAngel receives and demodulates the message correctly.

Under the AIS Demodulator Received Messages display, we see:

```
NMEA:    !AIVDM,1,1,,,>>O`Uh0lt:1@E=@,2*2c
Data:    Safety message: M
```

Offline decoders show that the full message is encoded in the AIVDM string displayed by SDRAngel. Also, commercial VHF receivers are able to decode and display the full message.

srcejon, 2024-04-29:
Any chance you can copy and paste the contents of the Hex column, please?

madmanguruman, 2024-04-29:
Thank you, Here is a fresh capture:

```
Date: Mon Apr 29 2024
Time: 13:37:43
MMSI: 972690000
Country: 
Data: Safety message: M
NMEA: !AIVDM,1,1,,,>>O`BD0lt:1@E0,4*5c
Hex:  38e7e8494034f0a05054
Slot: 1631
```

srcejon, 2024-05-16:
Should be fixed in 7.20.1

closed 2024-05-16 by srcejon