f4exb/sdrangel #2097: SSBModSource.m_audioFifo overflow with Simple PTT

alessiodilibertoeng, 2024-05-01:
Dear F4EXB and all developers,

first of all thank you for your great software, it is really amazing and powerful!

Now, as shown in debug log file attached to this post (sdrangel.SSBModSource.m_audioFifo.overflow.log), I apparently have some kind of issue when using SDRangel with Simple PTT feature.

Graphical setup as per the screenshot attached to this post. Essentially I am using 2 SDR boards: one for TX (HackRF one) and one for RX (RSP1 Chinese clone) and using such Simple PTT feature to switch from one board to the other board.

I am using them on Raspberry Pi 4B 8GB memory running on bullseye. SDRangel version is 7.18.0, compiled and installed with both feature 24BIT=OFF and 24BIT=ON (I have tried both configurations hoping that was the problem I am going to show, but unfortunately same issue...).

The main problem is that after some random switching from TX to RX and vice-versa, SDRangel enter in an infinite loop showing in the UX TERM (and also in the attached log file) an **issue with SSBModSource.m_audioFifo overflow** (please see attached log file for details). Therefore, because of that, I have to turn off the UX TERM in order to stop SDRangel.

Sometimes, instead of being in this indefinite infinite state, **it directly crashes**.

From my experimentation, it looks like that this behavior is more susceptible to DECIMATION factor in the RX side. The more I increase this factor, the more it is unstable and goes into this overflow issue. However, even for DECIMATION = 1 it does the same behavior sometimes, so it looks like that the problem is DECIMATION factor independent.

May I kindly ask you for your help about solving this issue? Is there any setting I have to do or you think it is a bug?

Thanks and regards,

Alessio, IZ0SAV

[image]
[sdrangel SSBModSource.m_audioFifo overflow.log](https://github.com/f4exb/sdrangel/files/15176453/sdrangel.SSBModSource.m_audioFifo.overflow.log)

f4exb, 2024-05-01:
Check the CPU usage. Your Rasp-Pi is probably getting overloaded.

alessiodilibertoeng, 2024-05-03:
I have checked the CPU usage and its peak is max 60% when SDRangel is on using this PTT feature. And SDRangel process is at max 26%.

Maybe some issue about SimplePTT handling the RX/TX switching?

The fact is that when I use either the RX or the TX, there is no problem at all. Just when using this PTT feature.

closed 2024-05-08 by alessiodilibertoeng