f4exb/sdrangel #2098: De-phasing with 2 antenna's

ThermoNuclear65, 2024-05-01:
Hi ,

is it at all possible the create a plug-in for the RSP-Duo where someone can de-phase the 2 antenna signals to get rid of the noise in software.

SDR Uno makes a good job out off it , but i just love using SDRAngel.

If not or to big workload I'll understand , but let me know please.

Kind regards , 73,

Patrick

srcejon, 2024-05-01:
Good idea, but let me expand the scope to supporting antenna diversity in general, so not just for noise cancellation, but also other forms such as selecting strongest signal / combining signals etc.

Rather than just working with MI devices like the Duo, would be interesting to also be able to use completely different receivers, although phase alignment might be tricky.

ThermoNuclear65, 2024-05-01:
Hi ,

With 2 separate receivers is even a better idea.Please give it a go at your convenience . I live in an urban area polluted with solar panels and cheap inverters.

Daytime is a disaster for noise suppression with just 1 antenna. 

Thanks for the quick and honest response.

Kind Regards, 73

Patrick

f4exb, 2024-05-01:
For a diversity setup to work correctly you need to have the receivers LOs and DACs synchronized to the same clock which concretely happens only on a few devices that are supported via the MIMO device plugins: BladeRF2, LimeSDR, Pluto+, XTRX. If the RSP duo is able to present a synchronized interleaved stream then it should be possible to support it this way. 

> would be interesting to also be able to use completely different receivers, although phase alignment might be tricky.

Yes tricky indeed, The most difficult part if even possible at all is to synchronize the USB streams. Therefore the above mentioned devices present the flow of samples by interleaving the samples of both receivers in the same data flow on the USB (or network). The Kraken SDR synchronizes multiple USB streams from several RTL-SDR devices using a common injected signal for synchronization. In addition to the hardware setup there is a fair amount of work to effectively synchronize the streams I haven't looked into this into detail.

For what exists now as MIMO channels the choice is fairly limited to basic interferometry and DOA but could be augmented with signal combination which output could be poured into a "normal" Rx channel using the local input for example.

srcejon, 2024-05-02:
> Yes tricky indeed, The most difficult part if even possible at all is to synchronize the USB streams. 

Yeah, there's two problems: the differing latency from each device and clock drift. While there will be some algorithms that require perfect phase sync to work and need a proper MIMO device, are there some simpler ones that don't? E.g. if we just want to continually select the strongest SSB voice signal from two different devices? Calculating the difference in latency should be fairly straightforward with cross-correlation, and as it's low quality voice, I'd guess that a little bit of phase difference when switching between the two isn't that important.

ThermoNuclear65, 2024-05-02:
Hi All ,

The RSP Duo seems to be able to do this , they have diversity control.
Maybe its worth looking at this one .

Kind Regards , 73

Patrick

f4exb, 2024-05-11:
Soon to be released. It works!
[image]

[image]

ThermoNuclear65, 2024-05-12:
Perfect , the best at work . Continua Please.

f4exb, 2024-05-13:
This is an example setup ...
[image]

It is using a Red Pitaya in "SDR transceiver compatible with HPSDR" mode that has its IN1 and IN2 inputs connected to Rx0 and Rx1 in the Metis MISO plugin (M:0). The M:0 spectrum on the top right shows the spectrum from the main antenna (dipole) connected to the IN2/Rx1 input which signal is affected by noise with peaks every 8 kHz. 

The IN1/Rx0 input is connected to an auxiliary wide loop antenna that picks up more of the noise. The Interferometer MIMO channel plugin (M0:0) is taking the Rx0 on channel A and Rx1 on channel B and phase shilft and gain is applied to B in order to eliminate the common noise input to A and B (phase difference of 6 degrees and gain of 1 dB). As you can see the noise has disappeared in the spectrum shown on the Interferometer A+B output. Here we also decimate by two to concentrate on the FT8 range.

The A+B output is sent to a Local Input device (R:1) which spectrum is shown on the bottom left and is identical to the one shown in the Interferometer window. Then we add a FT8 demodulator to the R:1 Rx chain (R1:0) to effectively decode the FT8 messages at 14.074 MHz which is the final result we want to obtain.

f4exb, 2024-05-13:
For RSP duo synchronous dual tuner mode support there is an interesting post here: https://sdrplayusers.net/forums/topic/rspduo-using-dual-tuners-with-api-3-06-in-c/

flat-linex, 2025-06-06:
Would this strategy work using two RTL-SDR blog v4 connected to two different antennas (without using Red Pitaya)?

f4exb, 2025-06-06:
You would need two things:

- have a common clock between RTL-SDRs
- synchronize the USB channels

While the first could be achieved with a hardware hack the latter is quite difficult to obtain in software if at all possible.

Radiomix2000, 2026-02-05:
USRP (B2x0 and some others) have 2 synchronous receivers. Any chance to have implemented diversity working with them?