f4exb/sdrangel #2099: Badly Distorted Audio Ubuntu 24.04

shull895, 2024-05-04:
After installing SDRAngel on Ubuntu 24.04, I found that audio is badly distorted. It is utterly unintelligible.  I have no sound issues with Ubuntu 24.04 and SDR++ . SDR++ works perfectly.  Anyone have a fix?

shull895, 2024-05-19:
Is nobody else using Ubuntu 24.04 with SDRangel 7.20? This is a POS. It does not provide inteligible audio. It installs without errors but audio is absolutely horrible. I tried self compilation as well as the snap version of it. Neither one works with Ubuntu 24.04 on ASUS X509ma AMD64 bit machine.

srcejon, 2024-05-19:
I don't have any problems with 24.04. If you want help, please include more information, such as screenshots showing the signal your are receiving and all of the settings you are using. Also, it may be helpful to include log files.

shull895, 2024-05-19:
Did you do a completely fresh install or just upgrade? I did a completely fresh installation of Ubuntu 24.04 for AMD64.
Is your computer a full computer or laptop? I have no problems using SDR++ on my laptop but SDRangel has horrible audio. Like its not giving me pulse audio but rather alsa.

srcejon, 2024-05-20:
A fresh install. Doesn't really matter what my computer is - it's more details from you we need to help. So please include screenshots and log files.

shull895, 2024-05-20:
What log files? How to get them, where located?

srcejon, 2024-05-20:
Under Menu Preferences > Logging

shull895, 2024-05-20:
Ya, I found that but where would the log file be located so I can send it.

srcejon, 2024-05-20:
Check log to file then press the file icon to set the location of the log file to whereever you want:

[image]

shull895, 2024-05-21:
Attached is a screenshot of my SDRangel screen and the log file.

image.png (view on web)<https://github.com/f4exb/sdrangel/assets/57259258/f2554df7-0b09-463c-83aa-b524e7b7eebf>

srcejon, 2024-05-21:
E-mail attachments don't work - you need to add via github webpage.

srcejon, 2024-05-21:
Nothing obvious. Does the Broadcast FM Demod work for regular FM radio?

shull895, 2024-05-21:
ALL audio is exactly the same. No matter whether it is AM, FM, NFM, WFM, all audio is the same. Badly distorted. This is the only program I have sound issues with. I use it just fine on my raspberry pi 4B with Ubuntu Noble but not on my laptop. All other programs work fine on the laptop.

srcejon, 2024-05-21:
What happens if you right click on the green audio icon in a demod window and change some of the audio settings? Try a different sample rate etc.

shull895, 2024-05-21:
What green audio icon? I don't see a green audio icon.

srcejon, 2024-05-21:
Look on the right hand side of the screenshot you uploaded.

shull895, 2024-05-21:
Oh, you mean the button that looks a speaker and is green when sound come out.

shull895, 2024-05-23:
I tried every audio device available from the right hand click on the green speaker with no success. I did try all the signal methods again, AM, NFM, Broadcast FM and noticed there is no WFM for those radios that still use 25KHz bandwidth like in Canada. Broadcast FM audio is fine but all others are still unintelligable audio. I also use this on my Raspberry Pi 4B on Ubuntu 24.04 and it works perfectly but not on my laptop. I also tried the Ubuntu snap version from Conical with the exact same results. Bad Audio.
I am at a loss for this problem.

shull895, 2024-05-23:
Channel spacing was 25 KHz and was reduced to 12.5 KHz 10 years or so ago and will be changing again to 6.25 KHz channel spacing in the near future. This is because so many radio licensing has congested the frequency spectrum. Canada still allows 25 KHz radios to be used as well as 12.5 KHz radios. That is why WFM (wide band FM) is still required.

Not sure what you mean by this. In the NFM Demod, you can set the RF BW to 25KHz. There's also the WFM Demod as well.

If that works, I'm at a loss as to why other demods dont. Does the volume setting make any difference?

f4exb, 2024-05-24:
I see you use HackRF and this guy has a pretty strong DC peak which kind of shows on your screenshot. Depending on the strength of the signal it may badly interfere with it. Turning on DC cancellation may or may not help since it will also cut off the central carrier which may be bad particularly on Naorrowband FM. So one thing you may want to try is to shift the NFM demod by a few kHz on the right or left off the center.

shull895, 2024-05-24:
Thank you for your reply Edouard, I have tried everything I can think of to correct this. I have this also on my RPI4B using Ubuntu 24.04 for RPI4b and have no issues. I also tried SDR++ on my laptop and it works perfectly on it. I am so frustrated with this I may just forget using it totally as I am not able to fix this. SDRangel does however work on Broadcast FM, perfectly. AM, NFM and anything that uses these methods, does not work correctly. If I can not understand the audio I am not going to use it.  I also had the same issue when I tried to install it on my Windows 11 Home.

shull895, 2024-05-31:
Good morning Edouard,

Just keep you in the loop, I wiped my Asus X509ma laptop, installed Debian 12 Bookworm, completed all the available updates and did a fresh installation of SDRAngel as well installed the driver for my HackRF SDR. I still have the same issue with extremely poor audio on AM, FM and Broadcast FM. In Debian 12 Bookworm the audio was bad in all three demod modes. I also wiped the laptop and installed Ubuntu 24.04 fresh installation and then installed SDRAngel and the HackRF driver. I had the exact same results. 

 

After looking at the log, I noticed that the SDRAngel software will not work with the Pipewire audio server which both Debian and Ubuntu have moved to for their sound server instead of Pulseaudio. I also installed GQRX and SDR++ to compare audio quality and understandability. BOTH of those pieces of software worked flawlessly with excellent voice quality. Would there be an issue with how SDRAngel processes receive audio to work with Pipewire?

 

Kind Regards

Steven Hull

f4exb, 2024-06-02:
It works on my RPi5 with Lubuntu 24.04 using the HDMI audio.

shull895, 2024-06-02:
I know, but that's not my issue.

f4exb, 2024-06-02:
you do need pulseaudio. What prevents you from installing it anyway?

shull895, 2024-06-02:
I don't like it.

f4exb, 2024-06-02:
I don't like pipewire...

f4exb, 2024-06-02:
pulseaudio is in the pre-requistes I think this is clear in the Wiki.

closed 2024-06-02 by f4exb

shull895, 2024-06-02:
Well Debian and Ubuntu don't use pulseaudio any more, they use pipewire.

f4exb, 2024-06-02:
From the Wiki...
[image]

Pulseaudio package in Ubuntu 24.04: https://launchpad.net/ubuntu/noble/+package/pulseaudio
And in Bookworm: https://packages.debian.org/bookworm/pulseaudio

shull895, 2024-06-02:
Tks. I'll try that.

From the Wiki...
image.png (view on web)<https://github.com/f4exb/sdrangel/assets/6192319/9eee2254-de0d-4f2c-86f9-69e19d722851>