f4exb/sdrangel #2108: Reverse API for Frequency Scanner?

sandric, 2024-05-10:
Hi, I'm playing with frequency scanner reverse api, the idea is to scan list of freq's and send reverse api request to check if it became "active" - e.g. power level got bigger than some theshold. I see that it does not send any change on any of channel changes I make, so I wonder if its my setup is wrong (its working for other channels I test though), feature is not implemented, or its not even planned to be implemented since this channel would generate a ton of requests each scan?
Thx

srcejon, 2024-05-10:
Currently it only reports via the GUI.

I could probably add the results to the channel report.

CharlieDeltaOskar, 2024-06-20:
Also, it would be great to have start/stop scanning controls.

closed 2024-11-01 by srcejon