f4exb/sdrangel #2115: ENABLE_SOAPYSDR=ON on macos

ask0n, 2024-05-16:
Hi, I'm building sdrangel on macos from sources with brew to enable SoapySDR support with `-DENABLE_SOAPYSDR=ON`
It builds but during the start it reports explicitly about:
```
(I) PluginManager::loadPluginsDir: Soapy SDR disabled skipping libinputsoapysdr.dylib
```

I'm particularly  interested to try to setup SoapyRemote device as described in [Wiki](https://github.com/f4exb/sdrangel/wiki/Soapy-Remote)
```
SoapySDRUtil --check=remote
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Loading modules... done
Checking driver 'remote'... PRESENT
```

I can confirm that remote driver works with `rtl_433` as expected.
What do I missing? Thanks!

srcejon, 2024-05-16:
Run sdrangel from the command line with --soapy

ask0n, 2024-05-17:
It shows now device in the list for `User arguments`, I can add `Arg string` with `addr=192.168.x.x` but when I try to use it it just fallbacks to File Input:
```
(D) PluginManager::registerSampleSource  SoapySDR Input  with source name  sdrangel.samplesource.soapysdrinput  and hardware id  "SoapySDR"
(D) PluginManager::registerSampleSink  SoapySDR Output  with sink name  sdrangel.samplesink.soapysdroutput  and hardware id  "SoapySDR"
(D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.soapysdrinput
(D) DeviceSoapySDRScan::scan: Lib Version: v0.7.1-g5838bc91
(D) DeviceSoapySDRScan::scan: API Version: v0.7.1
(D) DeviceSoapySDRScan::scan: ABI Version: v0.7
(D) DeviceSoapySDRScan::scan: Install root: /usr/local
(D) SoapySDROutputPlugin::enumOriginDevices: 0 SoapySDR devices
(D) DeviceEnumerator::enumerateDevices: sdrangel.samplesink.soapysdroutput
(D) DeviceEnumerator::addNonDiscoverableDevices: Rx: SoapySDR[0:0] user defined
(D) DeviceEnumerator::addNonDiscoverableDevices: Tx: SoapySDR[0:0] user defined
(D) MainWindow::sampleSourceCreate: deviceSetIndex: 0 deviceIndex: 8 hardwareId: "SoapySDR" sequence: 0 id: "" serial: "SoapySDR-0" displayedName: "SoapySDR[0:0] user defined"
(D) MainWindow::sampleSourceCreate: non existent device replaced by File Input

srcejon, 2024-05-17:
Are the args just an IP address? I haven't used it, but see here:

https://github.com/f4exb/sdrangel/wiki/Soapy-Remote

ask0n, 2024-05-17:
Thanks @srcejon,

I found other related issues, including this [one](https://github.com/f4exb/sdrangel/issues/1441), where @f4exb explains that SoapySDR should be considered a last resort.

I'm new to the topic, and SoapyRemote was easy to launch in my environment. I've also read that compared to `rtl_tcp`, it is more flexible and supports both RX and TX.

Since I'm just beginning to choose the right set of tools for remote signal processing, and the clarification I found is from 2022, my question is the SDRangel server on a remote host still the best approach? Will this solution work with `rtl_433`?

srcejon, 2024-05-17:
What SDR are you trying to use? What sort of computer is the server? Do you need TX? Do you need more than 8-bit IQ data? Do you only want to use rtl_433 or do you want to use SDRangel as well on the client?

I'm actually working on a few new features to SDRangel's Remote TCP server/client right now, that should make it compelling compared to the other options, but still needs a bit more work.

ask0n, 2024-05-17:
I'm currently experimenting with RTL-SDR and HackRF. Once I've covered the basic functionality, I plan to add LimeSDR to the setup since TX is needed. I don't have any special requirements for IQ data at this point. My main priorities are rtl_433 and URH. Ideally, I'd like to use SDRangel on the client, primarily for RX.

closed 2024-07-03 by github-actions[bot]