f4exb/sdrangel #2118: Flatpack UI unusable on openSUSE with 4K monitor

shidel, 2024-05-18:
I don't know if the program is working or not. I can't get it to seem to do anything. In part because I'm new to SDR. But, also in part to the UI being incomprehensible with multiple elements overwriting each other, text being cropped, etc.

[image]

srcejon, 2024-05-18:
What do you have for UI scaling under Preferences > Graphics menu.

shidel, 2024-05-19:
Originally, just 100%. It was just as bad at that setting and tiny. Turned it up to about 200%. A little on the big side. But everything still just messed up. Not better, not worse, just bigger.  I turned it back to 100% and restarted the program for theses screen shots...
[image]
[image]

shidel, 2024-05-19:
Using KDE, global scale settings, just 100%. 

[image]

shidel, 2024-05-19:
This issue has been resolved.

closed 2024-05-19 by shidel