f4exb/sdrangel #2119: Crash quitting sdrangel when the SID window is open

dforsi, 2024-05-19:
With the version  1836ae3a74f5b6c9e16c0627974be0883e762cd3 tagged v7.20.1 (ie without my modifications) if I close the SID window before closing sdrangel, then there is no crash.

Steps to reproduce the crash:
* start with an empty workspace
* add the SID feature
* close the main window of sdrangel
* crash

With the default compilation I get:
```
[...]
2024-05-19 23:07:48.310 (I) WebAPIServer::stop: stopped web API server at http://:8091
Segmentation fault (core dumped)
```

When compiling with ASAN I get this:
```
[...]
2024-05-19 22:59:36.684 (I) WebAPIServer::stop: stopped web API server at http://:8091
2024-05-19 22:59:36.685 (D) FeatureUISet::freeFeatures: destroying feature [sdrangel.feature.sid]
2024-05-19 22:59:36.710 (D) FeatureGUI::~FeatureGUI
2024-05-19 22:59:36.710 (D) FeatureGUI::~FeatureGUI: end
AddressSanitizer:DEADLYSIGNAL
=================================================================
==332940==ERROR: AddressSanitizer: stack-overflow on address 0x7ffc0b07dff8 (pc 0x7f0171998af2 bp 0x000000000001 sp 0x7ffc0b07e000 T0)
    #0 0x7f0171998af2 in QWidget::setAttribute(Qt::WidgetAttribute, bool) (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x198af2) (BuildId: dfefd27f084c0dd066215fc79825fceae604f481)
    #1 0x7f01719988c9 in QWidgetPrivate::createWinId() (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1988c9) (BuildId: dfefd27f084c0dd066215fc79825fceae604f481)
    #3 0x7f01719989d7 in QWidgetPrivate::createWinId() (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1989d7) (BuildId: dfefd27f084c0dd066215fc79825fceae604f481)
    #4 0x7f01719989d7 in QWidgetPrivate::createWinId() (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1989d7) (BuildId: dfefd27f084c0dd066215fc79825fceae604f481)
    #5 0x7f017199899c in QWidgetPrivate::createWinId() (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x19899c) (BuildId: dfefd27f084c0dd066215fc79825fceae604f481)
    #6 0x7f017199899c in QWidgetPrivate::createWinId() (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x19899c) (BuildId: dfefd27f084c0dd066215fc79825fceae604f481)
    #7 0x7f017199899c in QWidgetPrivate::createWinId() (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x19899c) (BuildId: dfefd27f084c0dd066215fc79825fceae604f481)
    #8 0x7f017199899c in QWidgetPrivate::createWinId() (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x19899c) (BuildId: dfefd27f084c0dd066215fc79825fceae604f481)
    #9 0x7f017199899c in QWidgetPrivate::createWinId() (/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x19899c) (BuildId: dfefd27f084c0dd066215fc79825fceae604f481)
[more lines omitted]
```

srcejon, 2024-05-20:
I can't reproduce this. What distribution and version of Linux are you using?

dforsi, 2024-05-21:
I'm using Debian testing and compiling with clang version 16.0.6 (27) on x86_64-pc-linux-gnu.

Today I learnt that we can get the outermost frames of a stack trace using a negative count, so this is my backtrace, omitting most of the lines that look duplicated:
```
(gdb) bt 5

#0  0x00007ffff2b98a05 in QWidget::setAttribute(Qt::WidgetAttribute, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007ffff2b988ca in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#2  0x00007ffff2b989d8 in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#3  0x00007ffff2b989d8 in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#4  0x00007ffff2b989d8 in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
(More stack frames follow...)

(gdb) bt -40

#174533 0x00007ffff2b9899d in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174534 0x00007ffff2b9899d in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174535 0x00007ffff2b9899d in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174536 0x00007ffff2b9899d in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174537 0x00007ffff2b989d8 in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174538 0x00007ffff2b989d8 in QWidgetPrivate::createWinId() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174539 0x00007ffff2b9a87b in QWidget::winId() const () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174540 0x00007fffe79e89fc in QGstreamerVideoWidgetControl::eventFilter(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5MultimediaGstTools.so.5
#174541 0x00007ffff1ecc4b3 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#174542 0x00007ffff2b62f22 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174543 0x00007ffff1ecc748 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#174544 0x00007ffff2b961b9 in QWidgetPrivate::setWinId(unsigned long long) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174545 0x00007ffff2b9ae89 in QWidgetPrivate::setParent_sys(QWidget*, QFlags<Qt::WindowType>) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174546 0x00007ffff2ba6d1e in QWidget::setParent(QWidget*, QFlags<Qt::WindowType>) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174547 0x00007fffc6a1c903 in  () at /lib/x86_64-linux-gnu/libQt5MultimediaWidgets.so.5
#174548 0x00007fffc6a1cad9 in QVideoWidget::~QVideoWidget() () at /lib/x86_64-linux-gnu/libQt5MultimediaWidgets.so.5
#174549 0x00007fffc6a1cb5d in QVideoWidget::~QVideoWidget() () at /lib/x86_64-linux-gnu/libQt5MultimediaWidgets.so.5
#174550 0x00007ffff1ef740e in QObjectPrivate::deleteChildren() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#174551 0x00007ffff2ba16a6 in QWidget::~QWidget() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174552 0x00007ffff2ba19dd in QWidget::~QWidget() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174553 0x00007ffff1ef740e in QObjectPrivate::deleteChildren() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#174554 0x00007ffff2ba16a6 in QWidget::~QWidget() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174555 0x00007ffff2d1805d in QSplitter::~QSplitter() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174556 0x00007ffff1ef740e in QObjectPrivate::deleteChildren() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#174557 0x00007ffff2ba16a6 in QWidget::~QWidget() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174558 0x00007ffff2ba19dd in QWidget::~QWidget() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174559 0x00007ffff1ef740e in QObjectPrivate::deleteChildren() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#174560 0x00007ffff2ba16a6 in QWidget::~QWidget() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#174561 0x00007ffff6d34c40 in RollupContents::~RollupContents() (this=0x51d0001aead0, this=<optimized out>) at /home/sdrangel/sdrgui/gui/rollupcontents.h:31
#174562 0x00007ffff6dae76c in FeatureGUI::~FeatureGUI() (this=0x51d0001aea80, this=<optimized out>) at /home/sdrangel/sdrgui/feature/featuregui.cpp:179
#174563 0x00007fffb76a73ca in SIDGUI::~SIDGUI() (this=0x51d0001aea80, this=<optimized out>) at /home/sdrangel/plugins/feature/sid/sidgui.cpp:297
#174564 0x00007fffb76a73f2 in SIDGUI::~SIDGUI() (this=0x51d0001aea80, this=<optimized out>) at /home/sdrangel/plugins/feature/sid/sidgui.cpp:297
#174565 0x00007fffb76a26b3 in SIDGUI::destroy() (this=0x51d0001aea80) at /home/sdrangel/plugins/feature/sid/sidgui.cpp:52
#174566 0x00007ffff715aa21 in FeatureUISet::freeFeatures() (this=0x50400050c890) at /home/sdrangel/sdrgui/feature/featureuiset.cpp:71
#174567 0x00007ffff6daea99 in FeatureUISet::~FeatureUISet() (this=0x50400050c890, this=<optimized out>) at /home/sdrangel/sdrgui/feature/featureuiset.cpp:38
#174568 0x00007ffff715a364 in FeatureUISet::~FeatureUISet() (this=0x50400050c890, this=<optimized out>) at /home/sdrangel/sdrgui/feature/featureuiset.cpp:39
#174569 0x00007ffff6df000b in MainWindow::removeAllFeatureSets() (this=0x7fffecc0e780) at /home/sdrangel/sdrgui/mainwindow.cpp:1207
#174570 0x00007ffff6d3b2ce in MainWindow::~MainWindow() (this=0x7fffecc0e780, this=<optimized out>) at /home/sdrangel/sdrgui/mainwindow.cpp:332
#174571 0x000055555555ec1e in runQtApplication(int, char**, qtwebapp::LoggerWithFile*) (argc=1, argv=0x7fffffffdb78, logger=0x50700002cb00) at /home/sdrangel/app/main.cpp:212
#174572 0x000055555555f4a4 in main(int, char**) (argc=1, argv=0x7fffffffdb78) at /home/sdrangel/app/main.cpp:248
(gdb) 
```

srcejon, 2024-05-22:
> I'm using Debian testing and compiling with clang version 16.0.6 (27) on x86_64-pc-linux-gnu.

Ok, I'm using gcc.

From the stack trace, probably something to do with the video player.

PR #2121 (2024-05-23): Fix crash when quitting sdrangel when the SID window is open
Fixes a stack exhaustion that happens in some machines/compiler combinations.

Fixes #2119.

I'm adding a `if (m_solarDynamicsObservatory)` as in the constructor.
commit db57a785 (2024-05-23): Fix crash when quitting sdrangel when the SID window is open

Fixes a stack exhaustion that happens in some machines/compiler combinations.
Fixes #2119
closed 2024-05-23 by f4exb