f4exb/sdrangel #2127: Mirics chipset based blue (dolphin) MSI.SDR receiver (libmiri) is initialized with wrong flavour.

Peter3579, 2024-05-26:
$ lsusb
Bus 003 Device 009: ID 1df7:2500 SDRplay RSP1

The following change fixed it for me:

```
diff --git a/plugins/samplesource/sdrplay/sdrplayinput.cpp b/plugins/samplesource/sdrplay/sdrplayinput.cpp
index 41d6f4437..a02537a8e 100644
--- a/plugins/samplesource/sdrplay/sdrplayinput.cpp
+++ b/plugins/samplesource/sdrplay/sdrplayinput.cpp
@@ -110,12 +110,6 @@ bool SDRPlayInput::openDevice()
         return false;
     }
 
-    if ((res = mirisdr_set_hw_flavour(m_dev, MIRISDR_HW_SDRPLAY)) < 0)
-    {
-        qCritical("SDRPlayInput::openDevice: failed to set HW flavour: %s", strerror(errno));
-        return false;
-    }
-
     char vendor[256];
     char product[256];
     char serial[256];
@@ -142,6 +136,12 @@ bool SDRPlayInput::openDevice()
         m_variant = SDRPlayRSP1;
     }
 
+    if ((res = mirisdr_set_hw_flavour(m_dev, (m_variant == SDRPlayRSP1)?  MIRISDR_HW_DEFAULT : MIRISDR_HW_SDRPLAY)) < 0)
+    {
+        qCritical("SDRPlayInput::openDevice: failed to set HW flavour: %s", strerror(errno));
+        return false;
+    }
+
     qDebug("SDRPlayInput::openDevice: m_variant: %d", (int) m_variant);
 
     return true;

```

Peter3579, 2024-05-31:
Well, I don't have a receiver stick from SDRPlay.com but a Chinese alternative. Could that make the difference? The Soapy driver allows to select the needed flavour. If you say your stick also identifies with `1df7:2500` but needs the `MIRISDR_HW_SDRPLAY`. then my patch does not work for you and maybe a lot of others.

f4exb, 2024-06-01:
I do tested with the blue "Dolphin" Chinese clone and it does identify as `1df7:2500` 
dmesg:
```
[137332.627053] usb 1-1.4.4: new high-speed USB device number 26 using xhci_hcd
[137332.751590] usb 1-1.4.4: New USB device found, idVendor=1df7, idProduct=2500, bcdDevice= 2.00
[137332.751604] usb 1-1.4.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```
lsusb:
```
Bus 001 Device 026: ID 1df7:2500 SDRplay RSP1 
```

Peter3579, 2024-06-01:
My functionality tests are on VHF and higher since I currently don't have a HF antenna. For instance, I can't receive any 100 MHz FM broadcast station with `MIRISDR_HW_SDRPLAY`. Just the noise floor is visible in the spectrum (and changes with the gain setting).

According to the `hw_switch_freq_plan` in `soft.c` the MSI2500 GPIO-bits are selecting (in general) different RF filter paths with `MIRISDR_HW_SDRPLAY` and `MIRISDR_HW_DEFAULT`, respectively. So it might be that it makes no difference at your test frequency. Or the circuit of the Chinese clone receiver has (silently) changed in the meantime.

f4exb, 2024-06-02:
OK... I did not really pay attention to the actual data. Indeed this seems to fix the problem.

closed 2024-06-02 by f4exb

Peter3579, 2024-06-07:
Hi Erik,
The issue was that the flavour `MIRISDR_HW_SDRPLAY` was set whilst `MIRISDR_HW_DEFAULT` was needed.

The `hw_switch_freq_plan` located in `libmiri/src/soft.c` sets, depending on the flavour, the strategy how the tuner is configured and the GPIO-pins for selecting the right "frontent" are set. The frontend selection is what you do manually with the DIP switches at you other clone. The GPIO-setting is the very last entry in the struct below. The corresponding frequency bands in MHz are coded the first entry. 

```
typedef struct
{
    uint32_t low_cut;
    int mode;
    int upconvert_mixer_on;
    int am_port;
    int lo_div;
    uint32_t band_select_word;
} hw_switch_freq_plan_t;

hw_switch_freq_plan_t hw_switch_freq_plan_default[] = {
        {0,    MIRISDR_MODE_AM,  MIRISDR_UPCONVERT_MIXER_ON, MIRISDR_AM_PORT2, 16, 0xf780},
        {12,   MIRISDR_MODE_AM,  MIRISDR_UPCONVERT_MIXER_ON, MIRISDR_AM_PORT2, 16, 0xff80},
        {30,   MIRISDR_MODE_AM,  MIRISDR_UPCONVERT_MIXER_ON, MIRISDR_AM_PORT2, 16, 0xf280},
        {50,   MIRISDR_MODE_VHF, 0, 0, 32, 0xf380},
        {108,  MIRISDR_MODE_B3,  0, 0, 16, 0xfa80},
        {250,  MIRISDR_MODE_B3,  0, 0, 16, 0xf680},
        {259,  6              ,  0, 0, 8,  0xf680},
        {330,  MIRISDR_MODE_B45, 0, 0, 4,  0xf380},
        {960,  MIRISDR_MODE_BL,  0, 0, 2,  0xfa80},
        {2400, -1, 0, 0, 0, 0x0000},
};

```

To answer your first question: It is the blue aluminium case china clone with the white dolphin on in.

commit ade3eedf (2024-06-02): SDRPlay v1: fix setting of hardware flavour. Fixes #2127