f4exb/sdrangel #2134: Support for KrakenRF SDR

jpettingale, 2024-05-29:
Are there plans to support the 5-channel KrakenRF as a MIMO device?

f4exb, 2024-05-29:
I have one of these device so probably some day... I haven't studied yet how the support library works. I hope there is an API to trigger the receivers synchronization and that we can get data in a consolidated synchronized stream.

jpettingale, 2024-05-29:
The Kraken developers have provided a GNU radio block, https://github.com/krakenrf/gr-krakensdr/tree/main. While it was written specifically for GNU radio, parts of the code can be repurposed. In particular, I believe that https://github.com/krakenrf/gr-krakensdr/blob/main/python/krakensdr/krakensdr_source.py provides most of the software for interfacing with the Kraken. One port is for controlling the Kraken and another is for receiving the IQ stream.

f4exb, 2024-05-30:
I read again the manual... and indeed everything goes through the intermediate of a RPi4 to which one connects via WiFi hotspot. Hence everything goes via network which could indeed simplify the interface vs a direct USB interface. Left to sort out if it is possible to connect directly to the RPi via ethernet on your home network while you're at home but that's another story.

jpettingale, 2024-05-30:
That is one way of doing it, but it requires a large “pipe” between the RPI4 and the PC doing the data processing/analysis.  The Heimdall DAQ server outputs a lot of data! The alternative is to run the Heimdall DAQ software on a PC and do the data processing/analysis on the same computer. This is the way I do it. In this case, the large “pipe” is internal to the computer, and the Heimdall DAQ server runs fine under Linux. This is why they provided the GNU radio block. I use GNU radio software for data processing/analysis, but I want to make sdrangel my “one-stop shop” for all my radio astronomy needs. Hence the request!

f4exb, 2024-05-30:
OK but you keep the client / server architecture using the network using the loopback device (127.0.0.1) instead of having the DAQ server actually remote. Is this correct? What is nice about this architecture is that you can have it locally or remote and this does not change anything but the IP address on the client side.

jpettingale, 2024-05-30:
Yes, that is correct, it works on either two separate PCs or on one, depending on your needs! Yes, only the IP address changes, the data stream remains the same.

alphafox02, 2024-06-01:
It’s not limited to a Pi. Run this script on an x86_64 22.04 and you’ll have HeIMDALL etc. 

wget https://raw.githubusercontent.com/krakenrf/krakensdr_docs/main/install_scripts/krakensdr_x86_install_doa.sh
sudo chmod +x krakensdr_x86_install_doa.sh
./krakensdr_x86_install_doa.sh

I run it on the same system has SDRAngel.